mantidproject/mantid #16994: Segfault starting MantidPlot built with Python3 on OS X.

quantumsteve, 2016-07-22:
### Expected behavior

After building Mantid against Python 3, MantidPlot should open (though with errors registering some algorithms).

### Actual behavior

MantidPlot segfaults. 

The cause of the crash is `PyBytes_AsString` returning NULL  [here](https://github.com/mantidproject/mantid/blob/a966d75b5eb2f08539f144f6eb4c7664ca342a23/Framework/PythonInterface/mantid/kernel/src/Environment/ErrorHandling.cpp#L27) and [here](https://github.com/mantidproject/mantid/blob/a966d75b5eb2f08539f144f6eb4c7664ca342a23/Framework/PythonInterface/mantid/kernel/src/Environment/ErrorHandling.cpp#L51).

The underlying cause is not clear to me.

### Steps to reproduce the behavior

Install additional dependencies mentioned in [this gist](https://gist.github.com/quantumsteve/07d8bbe38b576c991c16e79e70ddacfc).

Build Mantid with `-DPYTHON_EXECUTABLE=/usr/local/bin/python3` and either start MantidPlot or try importing mantid.simpleapi from a python terminal.

### Platforms affected

OS X.

quantumsteve, 2016-10-26:
@martyngigg Did you already address this?

martyngigg, 2016-10-26:
@quantumsteve I don't have easy access to a macOS machine so I haven't looked at it I'm afraid.

quantumsteve, 2016-11-03:
This appears to have been fixed. See, for example, http://builds.mantidproject.org/job/master_clean-osx-10.11-python3/102/console

martyngigg, 2016-11-04:
Agreed. The tests are now working so MantidPlot is started up without a problem.

closed 2016-11-04 by martyngigg