mantidproject/mantid #16999: BinMD nonsense

AndreiSavici, 2016-07-22:
### Expected behavior
For non-aligned axes, if a basis vector is replaced by the negative, I would expect to see the same data if I integrate in a symmetric interval around the origin
### Actual behavior
I am not getting any data if the projection vector is (-0.5,1,0), but I get the correct answer for (0.5,-1,0)
### Steps to reproduce the behavior
```
w=DgsReduction('CNCS_7860')
data=w[0]
SetUB(Workspace=data, a=5.975, b=5.975, c=17, gamma=120, u='-0.0032,0.02455,-1', v='-1,0.1343,-0.06489')
MD=ConvertToMD(InputWorkspace=data, QDimensions='Q3D', Q3DFrames='HKL', QConversionScales='HKL')

PP=Projection([1,0,0],[-0.5,1,0],[0,0,1])
SliceWrong=CutMD(MD,Projection=PP.createWorkspace(OutputWorkspace='proj_wsp'),PBins=(0.02, [-0.1,0.1], 0.09, [1.5,2]),NoPix=True)
sv1=plotSlice(SliceWrong,xydim=[0,2],normalization=2, colormin=0,colormax=1e-4,limits=[-3.6,3.6,-10,10])

PM=Projection([1,0,0],[0.5,-1,0],[0,0,1])
SliceGood=CutMD(MD,Projection=PM.createWorkspace(OutputWorkspace='proj_wsm'),PBins=(0.02, [-0.1,0.1], 0.09, [1.5,2]),NoPix=True)
sv2=plotSlice(SliceGood,xydim=[0,2],normalization=2, colormin=0,colormax=1e-4,limits=[-3.6,3.6,-10,10])

```
Instead of `CutMD`, one can use

```
BinMD(InputWorkspace='MD', AxisAligned=False, 
            BasisVector0="['zeta', 0, 0], r, 1, 0, 0, 0", 
            BasisVector1="['-0.50eta', 'eta', 0], r, -0.5, 1, 0, 0", 
            BasisVector2="[0, 0, 'xi'], r, 0, 0, 1, 0", 
            BasisVector3='DeltaE, DeltaE, 0, 0, 0, 1', 
            OutputExtents='-1.94106,1.79894,-0.1,0.1,-8.94824,4.01176,1.5,2', 
            OutputBins='187,1,144,1', 
            NormalizeBasisVectors=False, OutputWorkspace='sliceMinus')
            
BinMD(InputWorkspace='MD', AxisAligned=False, 
            BasisVector0="['zeta', 0, 0], r, 1, 0, 0, 0", 
            BasisVector1="['0.50eta', '-eta', 0], r, 0.5, -1, 0, 0", 
            BasisVector2="[0, 0, 'xi'], r, 0, 0, 1, 0", 
            BasisVector3='DeltaE, DeltaE, 0, 0, 0, 1', 
            OutputExtents='-1.94106,1.79894,-0.1,0.1,-8.94824,4.01176,1.5,2', 
            OutputBins='187,1,144,1', 
            NormalizeBasisVectors=False, OutputWorkspace='slicePlus')
```
### Platforms affected
All

samueljackson92, 2016-07-28:
Here's a smaller example which I think reproduces this issue. It seems to be that when performing a negative shear we're translating as well?

```python

mdws = CreateMDWorkspace(Dimensions=3, Extents='-10,10,-10,10,-10,10', Names='A,B,C', Units='U,U,U')
FakeMDEventData(InputWorkspace=mdws, PeakParams='100000,-5,-5,0,1')
FakeMDEventData(InputWorkspace=mdws, PeakParams='100000,0,0,0,1')
FakeMDEventData(InputWorkspace=mdws, PeakParams='100000,5,5,0,1')

binned_ws_good = BinMD(
    InputWorkspace=mdws, AxisAligned=False, 
    BasisVector0='a,unit,1,0,0',
    BasisVector1='b,unit,0.5,1,0',
    BasisVector2='c,unit,0,0,1', 
    NormalizeBasisVectors=False,
    Translation=[0,0,0], 
    OutputExtents=[-20,20,-20,20,-20,20], 
    OutputBins=[100, 100, 1] )

binned_ws_bad = BinMD(
    InputWorkspace=mdws, AxisAligned=False, 
    BasisVector0='a,unit,1,0,0',
    BasisVector1='b,unit,-0.5,1,0',
    BasisVector2='c,unit,0,0,1', 
    NormalizeBasisVectors=False,
    Translation=[0,0,0], 
    OutputExtents=[-20,20,-20,20,-20,20], 
    OutputBins=[100, 100, 1] )
```

Here's what it looks like in the slice viewer. Left is original, middle is with a positive shear, right with a negative shear.
[image]

samueljackson92, 2016-07-28:
Same script run on the 3.7.1 release looks more reasonable. I guess the changes causing this are recent.

[image]

samueljackson92, 2016-09-01:
I think I've tracked the root of this down. When calling ```CutMD``` in the above script, basis vectors for each of the dimensions are being passed to ```BinMD```. ```BinMD``` currently supports non-orthogonal basis vectors for "3 bases in a 4D world", **BUT** in the general case (in this case 4 bases in a 4D world) it makes the assumption that the bases must be orthogonal. See [here](https://github.com/mantidproject/mantid/blob/master/Framework/MDAlgorithms/src/SlicingAlgorithm.cpp#L956) in SlicingAlgorithm.

Running @AndreiSavici's script above but with the 4th basis vector removed I get something a little more reasonable:
```python
w=DgsReduction('CNCS_7860')
data=w[0]
SetUB(Workspace=data, a=5.975, b=5.975, c=17, gamma=120, u='-0.0032,0.02455,-1', v='-1,0.1343,-0.06489')
MD=ConvertToMD(InputWorkspace=data, QDimensions='Q3D', Q3DFrames='HKL', QConversionScales='HKL')

BinMD(InputWorkspace='MD', AxisAligned=False, 
            BasisVector0="['zeta', 0, 0], r, 1, 0, 0, 0", 
            BasisVector1="['-0.50eta', 'eta', 0], r, -0.5, 1, 0, 0", 
            BasisVector2="[0, 0, 'xi'], r, 0, 0, 1, 0", 
            OutputExtents='-1.94106,1.79894,-0.1,0.1,-8.94824,4.01176', 
            OutputBins='187,1,144', 
            NormalizeBasisVectors=False, OutputWorkspace='sliceMinus')

sv1=plotSlice('sliceMinus',xydim=[0,2],normalization=2, colormin=0,colormax=1e-4,limits=[-3.6,3.6,-10,10])

BinMD(InputWorkspace='MD', AxisAligned=False, 
            BasisVector0="['zeta', 0, 0], r, 1, 0, 0, 0", 
            BasisVector1="['0.50eta', '-eta', 0], r, 0.5, -1, 0, 0", 
            BasisVector2="[0, 0, 'xi'], r, 0, 0, 1, 0", 
            OutputExtents='-1.94106,1.79894,-0.1,0.1,-8.94824,4.01176', 
            OutputBins='187,1,144', 
            NormalizeBasisVectors=False, OutputWorkspace='slicePlus')
            
sv2=plotSlice('slicePlus',xydim=[0,2],normalization=2, colormin=0,colormax=1e-4,limits=[-3.6,3.6,-10,10])
```

Which produces the following images in the slice viewer
[image]

This can be reproduced a little easier by making an MDWorkspace and faking some data:
```python
from mantid.api import Projection

to_cut = CreateMDWorkspace(Dimensions=4, Extents=[-1,1,-1,1,-1,1,-10,10], Names="H,K,L,E", Units="U,U,U,V")
# Add two fake peaks so that we can see the effect of the basis transformation
FakeMDEventData(InputWorkspace='to_cut', PeakParams=[10000,-0.5,0,0,0,0.1])
FakeMDEventData(InputWorkspace='to_cut', PeakParams=[10000,0.5,0,0,0,0.1])

#Good slicing
projection = Projection([1,0,0], [0.5,-1,0], [0, 0, 1])
proj_ws = projection.createWorkspace()
#out_md = CutMD(to_cut, Projection=proj_ws, PBins=([0.1], [0.1], [0.1], [-5,5]), NoPix=True)
BinMD(InputWorkspace='to_cut', AxisAligned=False, 
BasisVector0='A, r, 1, 0, 0, 0', 
BasisVector1='B, r, 0.5, 1, 0, 0', 
BasisVector2='C, r, 0, 0, 1, 0', 
BasisVector3='E, V, 0, 0, 0, 1', 
OutputExtents='-1,1,-1,1,-1,1,-5,5', OutputBins='200,300,200,1', NormalizeBasisVectors=False, 
OutputWorkspace='sliced')

#Bad slicing
projection = Projection([1,0,0], [-0.5,1,0], [0, 0, 1])
proj_ws = projection.createWorkspace()
#out_md2 = CutMD(to_cut, Projection=proj_ws, PBins=([0.1], [0.1], [0.1], [-5,5]), NoPix=True)
BinMD(InputWorkspace='to_cut', AxisAligned=False, 
BasisVector0='A, r, 1, 0, 0, 0', 
BasisVector1='B, r, -0.5, 1, 0, 0', 
BasisVector2='C, r, 0, 0, 1, 0', 
BasisVector3='E, V, 0, 0, 0, 1', 
OutputExtents='-1,1,-1,1,-1,1,-5,5', OutputBins='200,300,200,1', NormalizeBasisVectors=False, 
OutputWorkspace='sliced2')
```

closed 2016-09-15 by peterfpeterson