mantidproject/mantid #17001: Many test failures with NeXus v4.4.2

quantumsteve, 2016-07-25:
### Expected behavior

After upgrading to NeXus v4.4.2, Mantid builds and all unit tests pass.

### Actual behavior

Many unit tests fail

```
The following tests FAILED:
	464 - PythonInterfaceAPI_MultipleFilePropertyTest (Failed)
	532 - PythonAlgorithms_LoadLogPropertyTableTest (Failed)
	572 - PythonAlgorithms_TOFTOFMergeRunsTest (Failed)
	575 - PythonAlgorithms_ExportSampleLogsToCSVFileTest (Failed)
	576 - PythonAlgorithms_ExportExperimentLogTest (Failed)
	675 - DataHandlingTest_GroupDetectors2Test (Failed)
	687 - DataHandlingTest_LoadDetectorInfoTest (Failed)
	708 - DataHandlingTest_LoadLLBTest (Failed)
	715 - DataHandlingTest_LoadMuonLogTest (Failed)
	716 - DataHandlingTest_LoadMuonNexus1Test (Failed)
	717 - DataHandlingTest_LoadMuonNexus2Test (Failed)
	723 - DataHandlingTest_LoadNexusTest (Failed)
	745 - DataHandlingTest_LoadTOFRawNexusTest (Failed)
	746 - DataHandlingTest_LoadTest (Failed)
	785 - DataHandlingTest_SaveNexusProcessedTest (Failed)
	786 - DataHandlingTest_SaveNexusTest (Failed)
	803 - DataHandlingTest_StartAndEndTimeFromNexusFileExtractorTest (Failed)
	812 - AlgorithmsTest_AlphaCalcTest (Failed)
	817 - AlgorithmsTest_ApplyDeadTimeCorrTest (Failed)
	820 - AlgorithmsTest_AsymmetryCalcTest (Failed)
	824 - AlgorithmsTest_CalMuonDeadTimeTest (Failed)
	983 - AlgorithmsTest_PhaseQuadMuonTest (Failed)
	984 - AlgorithmsTest_PlotAsymmetryByLogValueTest (Failed)
	1073 - WorkflowAlgorithmsTest_MuonProcessTest (Failed)
	1179 - CurveFittingTest_TabulatedFunctionTest (Failed)
	1482 - CustomInterfacesTest_ALCDataLoadingPresenterTest (Failed)
	1514 - MantidPlotSliceViewerTest.py (SEGFAULT)
Errors while running CTest
```

### Steps to reproduce the behavior

Build and install NeXus v4.4.2, build Mantid and run the unit tests. 

### Platforms affected

All

quantumsteve, 2016-07-25:
@zjttoefs @stuartcampbell @peterfpeterson

rosswhitfield, 2016-07-25:
I see the same thing but with a shorter list of failing tests. They all fail to read HDF4 files with `does not look like a HDF file`.

```
The following tests FAILED:
	464 - PythonInterfaceAPI_MultipleFilePropertyTest (Failed)
	532 - PythonAlgorithms_LoadLogPropertyTableTest (Failed)
	717 - DataHandlingTest_LoadMuonNexus2Test (Failed)
	723 - DataHandlingTest_LoadNexusTest (Failed)
	745 - DataHandlingTest_LoadTOFRawNexusTest (Failed)
	746 - DataHandlingTest_LoadTest (Failed)
	785 - DataHandlingTest_SaveNexusProcessedTest (Failed)
	786 - DataHandlingTest_SaveNexusTest (Failed)
	812 - AlgorithmsTest_AlphaCalcTest (Failed)
	817 - AlgorithmsTest_ApplyDeadTimeCorrTest (Failed)
	820 - AlgorithmsTest_AsymmetryCalcTest (Failed)
	983 - AlgorithmsTest_PhaseQuadMuonTest (Failed)
	984 - AlgorithmsTest_PlotAsymmetryByLogValueTest (Failed)
	1073 - WorkflowAlgorithmsTest_MuonProcessTest (Failed)
	1179 - CurveFittingTest_TabulatedFunctionTest (Failed)
	1482 - CustomInterfacesTest_ALCDataLoadingPresenterTest (Failed)
```

peterfpeterson, 2016-07-25:
Why would upgrading hdf5 break hdf4 code?

peterfpeterson, 2016-07-25:
I haven't run the full test-suite, but the ones I try fail with nexus 4.4.1 (don't ask), hdf5 1.8.15, and hdf 4.2.11 on fc23.

rosswhitfield, 2016-07-25:
So code that hasn't been touched in a long time. It's definitely due to change in NeXus and not Mantid.

peterfpeterson, 2016-07-27:
To make this more interesting, `argus0026287.nxs` opens fine with `nxbrowse` in nexus v4.4.1.

quantumsteve, 2016-07-31:
@peterfpeterson 
```
 $ nxbrowse ExternalData/Testing/Data/UnitTest/argus0026287.nxs
NXBrowse 4.4.1 Copyright (C) 2009-2014 NeXus Data Format
Built with readline support - use <TAB> to complete commands and paths
failed to determine filetype for ExternalData/Testing/Data/UnitTest/argus0026287.nxs  
NX_ERROR: Can't open ExternalData/Testing/Data/UnitTest/argus0026287.nxs
```

rosswhitfield, 2016-07-31:
No problem here. This is with `v4.4.2`
```
$ nxbrowse ExternalData/Testing/Data/UnitTest/argus0026287.nxs
NXBrowse 4.4.1 Copyright (C) 2009-2014 NeXus Data Format
Built with readline support - use <TAB> to complete commands and paths
    NeXus_version = 4.1.0
    HDF_version = NCSA HDF Version 4.2 Release 1, February 17, 2005
    file_name = c:\macs\nexus\argus0026287.nxs
    file_time = 2009-06-03 17:11:11+0100
    creator = ISIS: RIKEN-RAL
NX/> 
```

Maybe that's why you have more failing tests than us?

quantumsteve, 2016-07-31:
I found an error in the homebrew formula that prevented HDF4 support from being built. Here an updated list of failures on OS X.

```
	469 - PythonInterfaceAPI_MultipleFilePropertyTest (Failed)
	538 - PythonAlgorithms_LoadLogPropertyTableTest (Failed)
	579 - PythonAlgorithms_TOFTOFMergeRunsTest (Failed)
	582 - PythonAlgorithms_ExportSampleLogsToCSVFileTest (Failed)
	583 - PythonAlgorithms_ExportExperimentLogTest (Failed)
	724 - DataHandlingTest_LoadMuonNexus2Test (Failed)
	730 - DataHandlingTest_LoadNexusTest (Failed)
	752 - DataHandlingTest_LoadTOFRawNexusTest (Failed)
	753 - DataHandlingTest_LoadTest (Failed)
	792 - DataHandlingTest_SaveNexusProcessedTest (Failed)
	793 - DataHandlingTest_SaveNexusTest (Failed)
	819 - AlgorithmsTest_AlphaCalcTest (Failed)
	824 - AlgorithmsTest_ApplyDeadTimeCorrTest (Failed)
	827 - AlgorithmsTest_AsymmetryCalcTest (Failed)
	990 - AlgorithmsTest_PhaseQuadMuonTest (Failed)
	991 - AlgorithmsTest_PlotAsymmetryByLogValueTest (Failed)
	1080 - WorkflowAlgorithmsTest_MuonProcessTest (Failed)
	1186 - CurveFittingTest_TabulatedFunctionTest (Failed)
	1489 - CustomInterfacesTest_ALCDataLoadingPresenterTest (Failed)
```

quantumsteve, 2016-07-31:
The problematic commit is [c431b5cbe39c7f79872a0fe7dbf6dd783e764172](https://github.com/nexusformat/code/commit/c431b5cbe39c7f79872a0fe7dbf6dd783e764172#diff-41b1e1f6f3730345f771f1b5f4a22e93R1114), which changed from using `NXgetnextattr` to `NXgetnextattra`. 

For HDF4, [`NXgetnextattra` throws an exception](https://github.com/nexusformat/code/blob/ca865ae33087a9c7e7beb3da708333a5c80a7e2a/src/napi4.c#L1961) and Mantid reports an error.

quantumsteve, 2016-07-31:
This looks identical to [NeXus issue #426](https://github.com/nexusformat/code/issues/426)

@zjttoefs When do you expect this issue to be resolved?

rosswhitfield, 2016-08-15:
This is fixed as of nexusformat/code@d4a37b1874ac4849ec3a9ba85fa90754ea0e5feb

zjttoefs, 2016-08-15:
Sorry @quantumsteve, I was on leave and have been a good boy too many times in that period. I needed to drop a few things.
From @rosswhitfield it looks like it's now all working. Do you need a release or something like that?

peterfpeterson, 2016-08-16:
The test rpm @stuartcampbell made worked for me on fc23. All (unit) tests pass.

peterfpeterson, 2016-10-13:
This is resolved with nexus v4.4.3

closed 2016-10-13 by peterfpeterson