mantidproject/mantid #17028: Improvements to the Reflectometry GUI

raquelalvarezbanos, 2016-07-26:
This is an umbrella issue. Scientists have provided a design document that can be found [here](https://github.com/mantidproject/documents/blob/master/Design/Reflectometry/Mantid_SuperGUI_v1.1.pptx) for an improved version of the ISIS Reflectometry interface.

Sub-issues:
- [ ] https://github.com/mantidproject/mantid/issues/16734
- [ ] https://github.com/mantidproject/mantid/issues/16869
- [ ] https://github.com/mantidproject/mantid/issues/16983

raquelalvarezbanos, 2016-10-14:
Umbrella issue. 

#16983 should be done for release 3.9
#17029 should be done for release 3.10

Therefore, marking this against release 3.10

gemmaguest, 2017-05-25:
I'm closing this issue because it is now only dependent on #17029, and we already have another umbrella ticket for that (#15566).

closed 2017-05-25 by gemmaguest