mantidproject/mantid #17029: Reflectometry interface should be able to autoreduce the data

raquelalvarezbanos, 2016-07-26:
From Max:

> Please find attached the aut-reduction script I've been testing this week on INTER. It's mostly using functions and bits of code from the current ReflGUI/Polref GUI, so I think it's fairly self-explanatory. Of course it's only using defaults at the moment when RROA is called, but that can easily be extended, and it's still lacking all sorts of functionality.
It should work for you provided you pass it a previously created transmission workspace.

```
try:
    from mantidplot import *
except ImportError:
    canMantidPlot = False #

#import ui_refl_window
#import refl_save
#import refl_choose_col
#import refl_options
import glob
import string
import csv
import os
import re
import time
import string
from operator import itemgetter
import itertools
from PyQt4 import QtCore, QtGui
from mantid.simpleapi import *
from isis_reflectometry.quick import *
from isis_reflectometry.convert_to_wavelength import ConvertToWavelength
from isis_reflectometry import load_live_runs
from isis_reflectometry.combineMulti import *
import mantidqtpython
from mantid.api import Workspace, WorkspaceGroup, CatalogManager, AlgorithmManager

try:
    _fromUtf8 = QtCore.QString.fromUtf8
except AttributeError:
    _fromUtf8 = lambda s: s

canMantidPlot = True

def all_subdirs_of(b='.'):
  result = []
  for d in os.listdir(b):
    bd = os.path.join(b, d)
    if os.path.isdir(bd): result.append(bd)
  return result

def latest_run():  
    latest=all_subdirs_of(r'//isis/inst$/ndxinter/instrument/data/')
    newest = max(glob.iglob(max(latest)+'/*.nxs'), key=os.path.getctime)
    return int(string.rstrip(string.lstrip(os.path.basename(newest),'INTER'),'.nxs'))

def _populate_runs_list(RBno):
        """
        Populate the list at the right with names of runs and workspaces from the archives
        """
        # Clear existing
        #self.listMain.clear()

        if 1:

            # Use ICAT for a journal search based on the RB number

            active_session_id = None
            if CatalogManager.numberActiveSessions() == 0:
                # Execute the CatalogLoginDialog
                login_alg = CatalogLoginDialog()
                session_object = login_alg.getProperty("KeepAlive").value
                active_session_id = session_object.getPropertyValue("Session")

            # Fetch out an existing session id
            active_session_id = CatalogManager.getActiveSessions()[-1].getSessionId() # TODO. This might be another catalog session, but at present there is no way to tell.

            search_alg = AlgorithmManager.create('CatalogGetDataFiles')
            search_alg.initialize()
            search_alg.setChild(True) # Keeps the results table out of the ADS
            search_alg.setProperty('InvestigationId', str(RBno))
            search_alg.setProperty('Session', active_session_id)
            search_alg.setPropertyValue('OutputWorkspace', '_dummy')
            search_alg.execute()
            search_results = search_alg.getProperty('OutputWorkspace').value

            #self.__icat_file_map = {}
            #self.statusMain.clearMessage()
            runlist=[]
            for row in search_results:
                file_name = row['Name']
                file_id = row['Id']
                description = row['Description']
                run_number = re.search(r'[1-9]\d+', file_name).group()

                if bool(re.search('(raw)$', file_name, re.IGNORECASE)): # Filter to only display and map raw files.
                    title = (run_number + '~ ' + description).strip()
                    runlist.append(title)
                    #self.SampleText.__icat_file_map[title] = (file_id, run_number, file_name)
                    #self.listMain.addItem(title)
            #self.listMain.sortItems()
            return runlist
            #del search_results
            

def make_tuples(rlist):
    # sort runs into tuples: run number, title, theta
    tup = ()
    for idx in rlist:
        split_title = re.split("th=|~", idx)
        if len(split_title) != 3:
            split_title = re.split("~", idx)
            if len(split_title) != 2:
                logger.warning('cannot transfer ' +  idx+ ' title is not in the right form ')
            else:
                theta = 0
                split_title.append(theta) # Append a dummy theta value.
                tup = tup + (split_title,)
        else:
            tup = tup + (split_title,) # Tuple of lists containing (run number, title, theta)

    tupsort = sorted(tup, key=itemgetter(1, 2))
    print tupsort
    return tupsort

def sort_runs(tupsort):
    # sort tuples of runs into groups beloning to one sample title
    row = 0
    complete_list=[]
    for _key, group in itertools.groupby(tupsort, lambda x: x[1]): # now group by title
        col = 0
        run_angle_pairs_of_title = list() # for storing run_angle pairs all with the same title
        for object in group:  # loop over all with equal title
            one_sample=[]
            run_no = object[0]
            angle = object[-1]
            run_angle_pairs_of_title.append((run_no, angle))

        for angle_key, group in itertools.groupby(run_angle_pairs_of_title, lambda x: x[1]):
            runnumbers = "+".join(["%s" % pair[0] for pair in group])

            col = col + 5
            if col >= 11:
                col = 0
            else:
                one_sample.append((runnumbers,angle_key))
        row = row + 1
        print one_sample
        complete_list.append(one_sample)

    sortedList = sorted(complete_list, key=lambda runno: runno[0])
    return sortedList
###

def autoreduce(RBno,transRun, runRange=[], oldList=[], auto=0):
    rlist = _populate_runs_list(RBno) 
    tupsort = make_tuples(rlist)
    sortedList = sort_runs(tupsort)
    
    newList = [item for item in sortedList if item not in oldList]
    
    for sample in newList:
        wq_list=[]
        overlapLow=[]
        overlapHigh=[]
        for item in sample:
            runno = item[0]
            angle = item[1]
            runnos=string.split(runno,'+')
            runnos = [int(i) for i in runnos] # check if runs have been added together
            
            try:
                angle=float(angle)
            except ValueError:
                angle=0.0
                print "Could not determine theta! Skipping run."
                        
            if len(runRange) and not len(set(runRange) & set(runnos)):
                angle=0.0 # will be skipped below
            
            if float(angle)>0.0:
                ws = ConvertToWavelength.to_workspace(runno, ws_prefix="")
                if not mtd.doesExist(runno+'_IvsQ'):
                    wq, wlam, th = ReflectometryReductionOneAuto(InputWorkspace=ws, FirstTransmissionRun=transRun, thetaIn=angle, OutputWorkspace=runno+'_IvsQ', OutputWorkspaceWavelength=runno+'_IvsLam',)
                else:
                    wq=mtd[runno+'_IvsQ']
                    th=angle
                wq_list.append(runno+'_IvsQ')
                inst = wq.getInstrument()
                lmin = inst.getNumberParameter('LambdaMin')[0] + 1
                lmax = inst.getNumberParameter('LambdaMax')[0] - 2
                qmin = 4 * math.pi / lmax * math.sin(th * math.pi / 180)
                qmax = 4 * math.pi / lmin * math.sin(th * math.pi / 180)
                overlapLow.append(qmin)
                overlapHigh.append(qmax)
                dqq, two_theta = CalculateResolution(Workspace=wq, TwoTheta=angle)

        if len(wq_list):
            w1 = getWorkspace(wq_list[0])
            w2 = getWorkspace(wq_list[-1])
            Qmin = min(w1.readX(0))
            Qmax = max(w2.readX(0))

               
            #print Qmin, Qmax, dqq 
            #print overlapHigh
            if len(wq_list)>1:
                outputwksp = string.split(wq_list[0],sep='_')[0] + '_' + string.split(wq_list[-1],sep='_')[0][3:]
            else:
                outputwksp = string.split(wq_list[0],sep='_')[0] + '_IvsQ_binned'
            if not mtd.doesExist(outputwksp):
                wcomb = combineDataMulti(wq_list,outputwksp , overlapLow, overlapHigh,Qmin, Qmax, -dqq, 1, keep=True)
                SaveAscii(InputWorkspace=outputwksp,Filename='U:/CARDENAS/July2016/ascii_data/'+outputwksp+'.dat',Separator='Space',ColumnHeader=False,ScientificFormat=True,WriteSpectrumID=False)
            
    return sortedList

### run here:
start = latest_run()
print start
oldList=[]

while 1:
    now = latest_run()
    oldList=[]
    if now != start:
        oldList = autoreduce(1610149,'TRANS_38621_38622',range(38623,38700))
    time.sleep(5)
############

oldList = autoreduce(1610149,'TRANS_38621_38622',range(38623,38700))

```

raquelalvarezbanos, 2016-10-14:
I don't think we can get this done for release 3.9. But I would like it to be in release 3.10.

gemmaguest, 2017-03-02:
This is lower priority than other issues so will probably not be done in 3.10

gemmaguest, 2017-04-10:
We're prioritising this again because it is the final step to complete roadmap issue #15565

PranavBahuguna, 2017-04-28:
Due to the complexity of the task, I am splitting this down into multiple parts and making this an umbrella issue:

- Autoreducing a set of runs from an RB number
- Saving reduced runs
- Pausing and restarting autoreduction
- Reducing live data

gemmaguest, 2017-08-16:
Here is the lastest script sent by Max:

```
try:
    from mantidplot import *
except ImportError:
    canMantidPlot = False #

#import ui_refl_window
#import refl_save
#import refl_choose_col
#import refl_options
import glob
import string
import csv
import os
import re
import time
import string
from operator import itemgetter
import itertools
from PyQt4 import QtCore, QtGui
from mantid.simpleapi import *
from isis_reflectometry.quick import *
from isis_reflectometry.convert_to_wavelength import ConvertToWavelength
from isis_reflectometry import load_live_runs
from isis_reflectometry.combineMulti import *
import mantidqtpython
from mantid.api import Workspace, WorkspaceGroup, CatalogManager, AlgorithmManager

try:
    _fromUtf8 = QtCore.QString.fromUtf8
except AttributeError:
    _fromUtf8 = lambda s: s

canMantidPlot = True

def all_subdirs_of(b='.'):
  result = []
  for d in os.listdir(b):
    bd = os.path.join(b, d)
    if os.path.isdir(bd): result.append(bd)
  return result

def latest_run():  
    latest=all_subdirs_of(r'//isis/inst$/ndxinter/instrument/data/')
    newest = max(glob.iglob(max(latest)+'/*.nxs'), key=os.path.getctime)
    return int(string.rstrip(string.lstrip(os.path.basename(newest),'INTER'),'.nxs'))

def _populate_runs_list(RBno):
        """
        Populate the list at the right with names of runs and workspaces from the archives
        """
        # Clear existing
        #self.listMain.clear()

        if 1:

            # Use ICAT for a journal search based on the RB number

            active_session_id = None
            if CatalogManager.numberActiveSessions() == 0:
                # Execute the CatalogLoginDialog
                login_alg = CatalogLoginDialog()
                session_object = login_alg.getProperty("KeepAlive").value
                active_session_id = session_object.getPropertyValue("Session")

            # Fetch out an existing session id
            active_session_id = CatalogManager.getActiveSessions()[-1].getSessionId() # TODO. This might be another catalog session, but at present there is no way to tell.

[... 162 lines cut ...]
start = latest_run()
print start
oldList=[]

while 1:
    now = latest_run()
    oldList=[]
    if now != start:
        oldList = autoreduce(1710319,['TRANS','TRANS'])
    time.sleep(5)
############

#For the missing run for the time being:
run1='43760'
run2='43761'
Load(Filename=r'\\isis\inst$\ndxinter\instrument\data\cycle_17_1\INTER000'+run1+'.raw', OutputWorkspace=run1+'.raw')
ReflectometryReductionOneAuto(InputWorkspace=run1+'.raw', ThetaIn=0.80, WavelengthMin=2.8, FirstTransmissionRun='TRANS_SM', MomentumTransferMin=0.010327523398054171, MomentumTransferStep=0.055474542204445539, MomentumTransferMax=0.062389155094064494, ScaleFactor=1, OutputWorkspaceBinned='wq', OutputWorkspace=run1+'_IvsQ', OutputWorkspaceWavelength=run1+'_IvsLam')
SaveNexusProcessed(InputWorkspace=''+run1+'_IvsQ', Filename='U:/Lawrence/May2017//'+run1+'_IvsQ.nxs')
Load(Filename=r'\\isis\inst$\ndxinter\instrument\data\cycle_17_1\INTER000'+run2+'.raw', OutputWorkspace=run2+'.raw')
ReflectometryReductionOneAuto(InputWorkspace=''+run2+'.raw', ThetaIn=2.3, FirstTransmissionRun='TRANS', MomentumTransferMin=0.029696022073279912, MomentumTransferStep=0.055055827052456999, MomentumTransferMax=0.33476774783608887, ScaleFactor=1, OutputWorkspaceBinned='wq', OutputWorkspace=run2+'_IvsQ', OutputWorkspaceWavelength=run2+'_IvsLam')
CloneWorkspace(InputWorkspace=''+run1+'_IvsQ', OutputWorkspace='currentSum')
Stitch1D(LHSWorkspace='currentSum', RHSWorkspace=''+run2+'_IvsQ', OutputWorkspace='currentSum', StartOverlap=0.033620731892181156, EndOverlap=0.097474407070121408, Params='0.010328,-0.055056,0.33')
RenameWorkspace(InputWorkspace='currentSum', OutputWorkspace=run1+'_'+run2)
SaveNexusProcessed(InputWorkspace=run1+'_'+run2, Filename='U:/Lawrence/May2017//'+run1+'_'+run2+'.nxs')
```

gemmaguest, 2017-09-07:
Here's an update on what's left to do, plus an additional request from Max about processing a subset of runs:

- [ ] Don't autoreduce transmission runs (#20089)
- [ ] Make autoreduction button toggle between play/pause (#20071)
- [ ] Reduce live data (i.e poll to check for new runs)
- [ ] Save reduced runs (if a save path is specified on the ASCII tab, the results should automatically be saved to that path)
- [ ] Users should optionally be able to specify a subset of runs that they want to process.

PR #22443 (2018-06-21): Autoprocessing on reflectometry GUI
**Description of work**
This PR adds fully automatic processing of an investigation on the `ISIS Reflectometry` GUI.
- The `Autoprocess` button transfers all rows to the table and processes them.
- When all runs have been processed, a background processes is started to poll for new runs from ICat. If any are found, they are also transfered and processed.
- Some minor refactoring has been done to reduce complexity. The queues of items to process have been removed from `GenericDataProcessorPresenter` and instead the processed state of rows is used to skip rows [... cut]
closed 2018-06-21 by SimonHeybrock