mantidproject/mantid #17046: Add workflow diagram for the PEARL scripts

FedeMPouzols, 2016-07-27:
Add a workflow algorithm for these scripts: http://docs.mantidproject.org/nightly/api/python/techniques/PearlPowderDiffractionISIS-v1.html#pearl-powder-diffraction-ref
(similarly as it was done for the "CRY" scripts, although much simpler: http://docs.mantidproject.org/nightly/api/python/techniques/CryPowderDiffractionISIS-v1.html#cry-powder-diffraction-ref).

This requires going through the code, and understand what are the input/output files and workspaces and how these are produced by the sequence of algorithms. The inputs and outputs should be clearly visible, and it should also show where the absorption correction happens in the workflow.

DavidFair, 2016-08-04:
Initially just map out Focus algorithm to check core functionality

FedeMPouzols, 2016-08-04:
The core routine is `pearl_run_focus`: https://github.com/mantidproject/mantid/blob/master/scripts/PearlPowderISIS/pearl_routines.py#L343
Let's focus on this one for a first diagram.

FedeMPouzols, 2016-08-04:
Later on we could address `pearl_createvan`: https://github.com/mantidproject/mantid/blob/master/scripts/PearlPowderISIS/pearl_routines.py#L621.

PR #17158 (2016-08-11): Engg:: Create PEARL workflow for run_focus function
Description of work.

Adds a workflow diagram for the PEARL script function `pearl_run_focus`. 

This PR should pass the documentation tests and the below diagram should be clear and comprehensible
The latest version of this diagram is attached:

[image]

Fixes #17046 .

The release notes have been updated

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [ ] Does the code conform to the [... cut]
closed 2016-08-11 by peterfpeterson