mantidproject/mantid #17067: Extending the functionality of parameters in fit functions

jmborr, 2016-07-28:
A high priority here in SNS regarding QENS data has become global fitting, and in this context there's a non-trivial issue regarding the parameters of a fit function: Many a times the user wants to impose certain Q-dependence on the fit parameter rather than infer that dependence from a plot of the parameter versus Q, after the carrying out the fit.

One example: User wants a stretched exponential with the relaxation time following a power-law, \tau=A*Q^B. Now instead of \tau I have A and B as fit parameters, and are Q-independent by design.

In the previous example, we cannot reuse the already existing `StretchExp` fit function, Instead, one would have to code from a whole new fit function. A possible solution is to expose `StretchExp` in python and then _dynamically_ instantiate a new fit function class that will integrate `StretchExp` and the previous Q-dependence. This new fit function would have parameters A and B instead of \tau.

jmborr, 2016-07-28:
@mantid-roman @louisemccann @ElliotAOram @gvardany @martyngigg @stuartcampbell 
Dear all,
This is a design issue on the fit functions that is quite pressing for QENS data. Please take a look and share your ideas and comments about possible ways to solve.
Best,
.Jose

stuartcampbell, 2016-08-01:
@martyngigg Isn't this in all practical sense the same as the model fitting parts of TobyFit ?

martyngigg, 2016-08-02:
@stuartcampbell Yes I think so. There is some parametrized resolution portion convoluted with a specified user model. I don't think I see away around having to add/wrap `StretchExpFT` in some way though.

There are still many questions hanging over the TobyFit stuff. Not least would be that I'm not wedded to the particularly design as I think the way we wanted to go with being able to define the user model functions more simply in Python means somethings will have to change.

There is also the question of what we should be fitting. The current framework only understands MD events but should probably be able to accept histograms too.

jmborr, 2017-05-09:
Not feasible in current C++ fitting framework without major refactoring. Maybe makes more sense to extend a model with the python interface

closed 2017-05-09 by jmborr