mantidproject/mantid #17082: Out-of-bounds memory access in CompactMDTest

quantumsteve, 2016-07-29:
### Expected behavior

run `MDAlgorithmsTest_CompactMDTest` without out-of-bounds memory access.

### Actual behavior

MDHistoWorkspaceIterator::jumpTo is being passed a very large and out of bounds index. I traced it back to MDHistoWorkspaceIterator::jumpToNearest, which in one case returns a negative value. 

I also get the warning below. Not sure if it is relevant. 
```
IntegrateMDHistoWorkspace-[Warning] Rounding min from: -4 to the nearest whole width at: -5
IntegrateMDHistoWorkspace-[Warning] Rounding max from: 4 to the nearest whole width at: 5
```

I believe there is a logic error in either `IntegrateMDHistoWorkspace` or `CompactMD`, but having trouble tracking it down.

### Steps to reproduce the behavior

run MDAlgorithmsTest_CompactMDTest while checking [here](https://github.com/mantidproject/mantid/blob/master/Framework/DataObjects/src/MDHistoWorkspaceIterator.cpp#L279) if `dExact` < 0.0.

### Platforms affected

All

quantumsteve, 2016-07-29:
I'm marking this high priority since @VickieLynch has users reporting intermittent segfaults when working with MDWorkspaces.

samueljackson92, 2016-08-31:
The negative index is appearing in ```test_compact_md_throws_when_loading_empty_workspace```  which checks running the algorithm on a workspace with no signal present. In that case, CompactMD will (unlike what the name implies) not throw an exception, but set the min/max bins to the upper and lower bounds of the dimension. ```setMinMaxBins``` in ```IntegrateMDHistoWorkspace``` is then taking those values and snapping to the ```min-1``` and ```max+1``` (and output the warning listed above) which I believe is then causing a negative index.

The solution here might just be to check that the values snapped to are within the limits of the dimension?

OwenArnold, 2016-08-31:
This is the Horace analogue of what `CompactMD` should be doing http://horace.isis.rl.ac.uk/Reshaping_etc#compact

OwenArnold, 2016-08-31:
I would use that description to set the behaviour.

samueljackson92, 2016-08-31:
@OwenArnold that doesn't appear to mention what we should do in the case where your workspace has no signal? Currently in that case we appear to set the bin limits to the min/max so you'd actually end up not cropping anything if the dimension doesn't contain any data.

samueljackson92, 2016-08-31:
@OwenArnold Looking at the git history it appears to have been decided that that behaviour is correct:
https://github.com/mantidproject/mantid/issues/14032

samueljackson92, 2016-09-01:
@OwenArnold @quantumsteve I'm wondering if it's worth putting a range check and a runtime exception in ```jumpToNearest```? Although the ```jumpTo``` documentation states:
```cpp
//----------------------------------------------------------------------------------------------
/** Jump to the index^th cell.
 *  No range checking is performed, for performance reasons!
 *
 * @param index :: point to jump to. Must be 0 <= index < getDataSize().
 */
```
So I don't know if that would have a negative effect on performance..

quantumsteve, 2016-09-01:
@samueljackson92 I think a range check would be a good idea. We'd want to measure the performance difference ( Are we limited by compute or memory bandwidth?). Otherwise a Debug-only assertion would be an alternative.  

Is jumping to the last point (one past the end of the signal array) valid? If so, one could pass the check and still do an out-of-bounds memory access to the signal array or mask array. We may want to add assertions to `getIsMasked()` and `getSignal`.

samueljackson92, 2016-09-02:
I think it'll be computationally limited (iterating over ```x``` bins with ```n``` dimensions). There's a performance test already there. I'll experiment with inserting a bounds check and evaluate the performance difference.

OwenArnold, 2016-09-02:
@samueljackson92 Yes I was going to suggest doing just that. I expected that the range check would be a significant overhead, but lets see what the performance tests come back with.

samueljackson92, 2016-09-02:
@quantumsteve @OwenArnold Having a look at this we probably don't want to put a check in the ```jumpTo``` method. It doesn't actually have a huge impact on performance test time, but that method is called ~2 million times and adding a check (unsurprisingly) doubles the execution time of the method.

But the ```jumpToNearest``` method is only called once per parallel thread in ```IntegrateMDHistoWorkspace```, and is the only place (currently) that the method is being used. So perhaps it would be acceptable to put a check in there?

PR #17387 (2016-09-16): Fix CompactMD out of bounds bug
This fixes the case where an invalid index was being returned if the min/max given to ```IntegrateMDHistoWorkspace``` was on the edge of the space. If the "snapped" bin value is outside of the range of the dimension we now force it to be the very edge of the dimension.

**To test:**
 - Code review
 - Check tests for ```IntegrateMDHistoWorkspace``` and ```CompactMD``` are fine.
 - Check CompactMD for the case where the workspace is all zero and the case where there is data on the limits of the space. Here's a script to get you started:

```python
import math
#cre [... cut]
closed 2016-09-16 by SimonHeybrock