mantidproject/mantid #17084: Invalid indices generated while using the Slice Viewer.

quantumsteve, 2016-07-31:
### Expected behavior

The code comment [here](https://github.com/mantidproject/mantid/blob/966c761c696da7845cedc4fe94af83758612dd9d/Framework/DataObjects/src/MDHistoWorkspace.cpp#L622) says so invalid indices should never occur.

### Actual behavior

In PR #17083, I found that an invalid index ( `= size_t(-1)` ) is being generated. I added checks to avoid the out-of-bounds memory accesses, but the larger issue still exists.

### Steps to reproduce the behavior

Check for invalid values of `linearIndex` while using the Slice Viewer or running `SliceViewerMantidPlotTest_SliceViewerPythonInterfaceTest`.

### Platforms affected

All

This was found in issue #16583.

quantumsteve, 2016-07-31:
@matthew-d-jones Would you mind looking into this?

matthew-d-jones, 2016-08-01:
@quantumsteve Sorry, I'm not working on Mantid now, but whoever works on this issue is welcome to talk with me about it on Slack, or in person if they are at ISIS.

quantumsteve, 2016-08-01:
@NickDraper Please assign as appropriate.

AntonPiccardoSelg, 2017-10-05:
@Mantid-Matthew Might be good to speak to Matthew Jones about this

Mantid-Matthew, 2020-09-01:
Refers to old version of slice viewer.

closed 2020-09-01 by Mantid-Matthew