mantidproject/mantid #17085: Add additional input property to ReflectometryReductionOne to apply flood corrections.

PranavBahuguna, 2016-08-01:
This is an issue split off from task 5 of issue #16455. 

We need an additional input property to apply flood corrections. This property was suggested to be called `DetectorEfficiencyCorrection`. This step should replace the two boxes `Multidetector?` and `Perform Direct Beam Normalization`. This will be a workspace which can contain:

- A single data point, in this case this value should be used as a normalisation constant shared across the different spectra in the input ws.

- A list of spectra with a single value: in this case each spectrum will have its own normalisation constant.

PranavBahuguna, 2016-08-05:
@raquelalvarezbanos For the `DetectorEfficiencyCorrection` workspace parameter, if a value of zero is provided for one of the spectra, it will attempt to divide the detector workspace spectra value by zero, resulting in an output value of `nan` or `inf`. Should I allow this to happen, or should I throw an error if a zero is found in the DEC workspace?

PranavBahuguna, 2016-08-08:
@raquelalvarezbanos Furthermore, in order for normalization to be carried out, the `DetectorEfficiencyCorrection` workspace must have the same x-units as the detector workspace (`detectorWS`), which is in wavelength. Should we throw an error if the `DetectorEfficiencyCorrection` workspace is provided with non-wavelength x-units, or should we simply convert it to those units?

raquelalvarezbanos, 2016-08-10:
Hi @PranavBahuguna in both cases (division by 0) and different X units, I would say we should throw an exception.

raquelalvarezbanos, 2017-02-13:
This will be fixed in https://github.com/mantidproject/mantid/issues/12524 and needs some additional discussion, hence I am closing this one.

closed 2017-02-13 by raquelalvarezbanos