mantidproject/mantid #17108: Abragam function parameters

tom-perkins, 2016-08-02:
See #17054:

The documentation for the `Abragam` fitting function is not clear about what some of the parameters represent, and the form of the function is different to some expressions seen elsewhere (notably, it is different to WiMDA). 

To ask scientists at the next muon meeting:
- Confirm what the parameters are, in particular `sigma` and `tau` 
- Confirm the scientists are happy with the function's form and the difference to how it is implemented elsewhere

Once we have this information, update the documentation accordingly.

tom-perkins, 2016-08-04:
Notes: 
- The function in the documentation does not match the code exactly. What we have is:
[image]

- Unclear whether it is supposed to be for transverse or longitudinal field
- `sigma` is not the decay rate (likely the width of local fields? `Delta` in other functions) and `tau` is probably the muon lifetime

tom-perkins, 2016-10-14:
To do at next muon meeting. Time estimate to update docs is < 1 hour once we have the answers to these questions from the scientists.

Anders-Markvardsen, 2017-01-09:
Contact Francis Pratt - recommendation from a muon meeting

closed 2021-03-02 by stale[bot]