mantidproject/mantid #17146: Test failures on Fedora 24

quantumsteve, 2016-08-04:
### Expected behavior

Mantid builds on Fedora 24 and passes all tests 

### Actual behavior

EqualBinSizesValidatorTest failed to build
```
[1592/3133] Building CXX object Framework/API/test/CMakeFiles/APITest.dir/EqualBinSizesValidatorTest.cpp.o
FAILED: /usr/bin/c++   -DBOOST_ALL_DYN_LINK -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_POSIX_TIME_STD_CONFIG -DENABLE_OPENCASCADE -DHAVE_STDINT_H -isystem /usr/include/nexus -I../Framework/Kernel/inc -I../Framework/HistogramData/inc -I../Framework/Geometry/inc -I../Framework/API/inc -isystem ../Testing/Tools/cxxtest -isystem ../Testing/Tools/gmock-1.7.0/include -isystem ../Testing/Tools/gmock-1.7.0/gtest/include -I../Framework/API/test/../../TestHelpers/inc -I../Framework/API/test/../../Nexus/inc -fopenmp -Wsuggest-override -Wall -Wextra -Wconversion -Winit-self -Wpointer-arith -Wcast-qual -Wcast-align -fno-common -Wno-deprecated -Wno-write-strings -Wno-unused-result -pedantic -Woverloaded-virtual -fno-operator-names -O3 -DNDEBUG -fvisibility=hidden   -std=gnu++14 -MMD -MT Framework/API/test/CMakeFiles/APITest.dir/EqualBinSizesValidatorTest.cpp.o -MF Framework/API/test/CMakeFiles/APITest.dir/EqualBinSizesValidatorTest.cpp.o.d -o Framework/API/test/CMakeFiles/APITest.dir/EqualBinSizesValidatorTest.cpp.o -c Framework/API/test/EqualBinSizesValidatorTest.cpp
In file included from ../Framework/API/inc/MantidAPI/MatrixWorkspaceValidator.h:6:0,
                 from ../Framework/API/inc/MantidAPI/EqualBinSizesValidator.h:5,
                 from /home/svh/Documents/mantid/Framework/API/test/EqualBinSizesValidatorTest.h:6,
                 from Framework/API/test/EqualBinSizesValidatorTest.cpp:15:
../Framework/Kernel/inc/MantidKernel/TypedValidator.h: In member function ‘virtual void TestDescription_suite_EqualBinSizesValidatorTest_test_null::runTest()’:
../Framework/Kernel/inc/MantidKernel/TypedValidator.h:135:7: internal compiler error: in emit_move_insn, at expr.c:3546
       throw std::invalid_argument("DataItem \"" + data->name() +
       ^~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://bugzilla.redhat.com/bugzilla> for instructions.
Preprocessed source stored into /tmp/ccG6VIh8.out file, please attach this to your bugreport.
```

The following tests FAILED:
```
	 74 - KernelTest_NDRandomNumberGeneratorTest (Failed)
	162 - GeometryTest_BraggScattererInCrystalStructureTest (Failed)
	267 - GeometryTest_StructureFactorCalculatorTest (Failed)
	313 - APITest_FunctionParameterDecoratorTest (Failed)
	327 - APITest_ILatticeFunctionTest (Failed)
	380 - APITest_MatrixWorkspaceTest (Failed)
	619 - DataObjectsTest_MDBoxIteratorTest (Failed)
	628 - DataObjectsTest_MDGridBoxTest (Failed)
	1093 - CurveFittingTest_FitTest (Failed)
	1218 - CrystalTest_CompositeClusterTest (Failed)
	1219 - CrystalTest_ConnectedComponentLabelingTest (Failed)
	1243 - CrystalTest_OptimizeCrystalPlacementTest (Failed)
	1347 - PSISINQTest_PoldiSpectrumPawleyFunctionTest (Failed)
	1356 - MDAlgorithmsTest_BooleanBinaryOperationMDTest (Failed)
	1460 - MDAlgorithmsTest_UnaryOperationMDTest (Failed)
	1464 - MDAlgorithmsTest_BinaryOperationMDTest (Failed)
	1481 - MantidWidgetsTest_DataProcessorCommandsTest (Failed)
	1484 - MantidWidgetsTest_GenericDataProcessorPresenterTest (Failed)
	1489 - CustomInterfacesTest_ALCBaselineModellingPresenterTest (Failed)
	1490 - CustomInterfacesTest_ALCDataLoadingPresenterTest (Failed)
	1493 - CustomInterfacesTest_ALCPeakFittingPresenterTest (Failed)
	1494 - CustomInterfacesTest_EnggDiffFittingPresenterTest (Failed)
	1495 - CustomInterfacesTest_EnggDiffractionPresenterTest (Failed)
	1497 - CustomInterfacesTest_ImageROIPresenterTest (Failed)
	1498 - CustomInterfacesTest_ImggFormatsConvertPresenterTest (Failed)
	1501 - CustomInterfacesTest_ReflLegacyTransferStrategyTest (Failed)
	1502 - CustomInterfacesTest_ReflMainWindowPresenterTest (Failed)
	1503 - CustomInterfacesTest_ReflMeasureTransferStrategyTest (Failed)
	1505 - CustomInterfacesTest_ReflRunsTabPresenterTest (Failed)
	1508 - CustomInterfacesTest_TomographyIfacePresenterTest (Failed)
	1512 - SliceViewerTest_CompositePeaksPresenterTest (Failed)
	1513 - SliceViewerTest_ConcretePeaksPresenterTest (Failed)
```

### Steps to reproduce the behavior

Build Mantid on Fedora 24

### Platforms affected

Fedora 24. Likely all system with GCC 6.1

quantumsteve, 2016-08-05:
The problematic test is [EqualBinSizesValidatorTest::test_null](https://github.com/mantidproject/mantid/blob/9c2023571d8a97c79828fe77777aa81051105f4d/Framework/API/test/EqualBinSizesValidatorTest.h#L20). Commenting out this test makes the internal compiler error go away.

The ice also goes away if we avoid [obtaining the address of a nullptr](https://github.com/mantidproject/mantid/blob/9c2023571d8a97c79828fe77777aa81051105f4d/Framework/Kernel/inc/MantidKernel/IValidator.h#L146)

The ice also goes away if I remote `data->name()` from [this line](https://github.com/mantidproject/mantid/blob/9c2023571d8a97c79828fe77777aa81051105f4d/Framework/Kernel/inc/MantidKernel/TypedValidator.h#L135)

rosswhitfield, 2016-08-05:
What I have discovered using gcc 6 is that the tests will pass if building with `-O0` but fail with `-O1` and up. If you do find any real problems with gcc 6 we should report the bugs to gcc instead of just finding workarounds.

quantumsteve, 2016-08-05:
@rosswhitfield I agree that we should try submitting these issues upstream, ideally with self-contained repros. Unfortunately, that's a lot easier said than done and I also suspect some of the test failures are undefined behavior that GCC no longer allows.

quantumsteve, 2016-08-10:
We are now down to 2 failing tests, same as other systems with GSL v2.

quantumsteve, 2016-08-14:
I created an (almost) self-contained repro that still produces the compiler error.
[gcc6_ice.cpp.txt](https://github.com/mantidproject/mantid/files/417068/gcc6_ice.cpp.txt)

g++ -O3 -DNDEBUG -std=gnu++1z -o gcc6_ice.cpp.o -c gcc6_ice.cpp

When I remove `boost::any` with `std::experimental::any`, the issue goes away 😞 
[gcc6_builds.cpp.txt](https://github.com/mantidproject/mantid/files/417069/gcc6_builds.cpp.txt)

g++ -O3 -DNDEBUG -std=gnu++1z -o gcc6_builds.cpp.o -c gcc6_builds.cpp

quantumsteve, 2016-08-15:
https://bugzilla.redhat.com/show_bug.cgi?id=1367212

quantumsteve, 2016-08-16:
GCC patch
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=77259

The failing unit tests are [temporary being skipped](https://github.com/mantidproject/mantid/pull/17245) and will be addressed in separate issues.

closed 2016-08-16 by rosswhitfield