mantidproject/mantid #17155: cppcheck 1.75

quantumsteve, 2016-08-08:
### Expected behavior

Mantid generates no cppcheck warnings.

### Actual behavior

 [cppcheck_with_includes.xml](https://github.com/mantidproject/mantid/files/405822/cppcheck_with_includes.xml.txt)
[cppcheck.xml](https://github.com/mantidproject/mantid/files/405821/cppcheck.xml.txt)

### Steps to reproduce the behavior

run cppcheck 1.75 with and without `CPPCHECK_USE_INCLUDE_DIRS`

### Platforms affected

all

quantumsteve, 2016-08-08:
@NickDraper @peterfpeterson Please split up and assign as appropriate. Keeping current allows us to use the homebrew formula instead of forking and maintaining our own version. Those that only appear with `CPPCHECK_USE_INCLUDE_DIRS` are probably better left for a maintenance period.

peterfpeterson, 2016-08-09:
I added this to the list of [maintenance tasks](https://github.com/mantidproject/documents/blob/master/Project-Management/TechnicalSteeringCommittee/reports/MaintenanceTasks.md) as it seems appropriate.

quantumsteve, 2016-11-21:
@NickDraper @peterfpeterson I think this has been largely taken care of, which the remaining issues best addressed in the maintenance period.

NickDraper, 2016-11-21:
Good, closed

closed 2016-11-21 by NickDraper