mantidproject/mantid #17162: ReflectometryReductionOne remove rebin in wavelength

PranavBahuguna, 2016-08-08:
This is split off from task 2 of #16455. Rebinning the detector workspace in wavelength is unwanted and should be removed (though `CropWorkspace` is fine). In addition, the `WavelengthStep` property is used solely with this portion of the code so it should be removed as well.

PranavBahuguna, 2016-08-09:
@raquelalvarezbanos One of the system tests (OFFSPECReflRedOneAuto) makes use of the `wavelengthStep` parameter that has now been removed. The result is then compared to "OFFSPECReflRedOneAuto_good_v3.nxs". However without the wavelength rebin, the result from the system no longer matches this file. Should I edit the file so it does match without wavelength rebinning, or should I remove the test altogether if it no longer serves any purpose?

raquelalvarezbanos, 2016-08-10:
@PranavBahuguna I would say that tests should not be removed unless there's a good reason to do so. In this case I think that what we should do is replace the reference file.

PR #17239 (2016-09-07): RRO remove rebin in wavelength
Description of work.

In the ReflectometryReductionOne algorithm, rebinning the detector workspace in wavelength was unwanted and the code necessary for doing this was to be removed. In addition, the `WavelengthStep` property was used solely with this portion of the code, so this property was to be removed as well as it no longer served any purpose.

Fixes #17162.

**Testing Instructions:**

Simply ensure that all unit and system tests work as before.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Indivi [... cut]
closed 2016-09-07 by FedeMPouzols