mantidproject/mantid #17165: segfault when algorithm dialog opens when MD workspace is present

rosswhitfield, 2016-08-08:
The problem doesn't exist at d1be85cb95a75b9ad7b3b0ae4255e861e773499b but does exist at 1e56170f874b7d5b1ff5d58e66d89e77c8d6ed0f therefore was introduced with #16967 

To reproduce:
* Load a MD workspace `mdws = LoadMD('MAPS_MDEW.nxs')`
* Open an algorithm dialog which **doesn't** take a MD workspace as input. _e.g._ `Plus`

Result is:
```
MantidApplication-[Fatal] Unexpected exception: basic_string::_M_create
MantidApplication-[Fatal] Continue working.
tcmalloc: large alloc 7021238724389208064 bytes == (nil) @ 
MantidApplication-[Fatal] Unexpected exception: std::bad_alloc
MantidApplication-[Fatal] Continue working.
bin/launch_mantidplot.sh: line 49: 23857 Segmentation fault      (core dumped) LD_PRELOAD=${LOCAL_PRELOAD} TCMALLOC_RELEASE_RATE=${TCM_RELEASE} TCMALLOC_LARGE_ALLOC_REPORT_THRESHOLD=${TCM_REPORT} LD_LIBRARY_PATH=${LOCAL_LDPATH} QT_API=pyqt $VGLRUN$INSTALLDIR/MantidPlot $*
```

MD algorithms work fine _e.g._ `PlusMD`

It looks like MD workspaces are being included in the list of possible InputWorkspaces when they shouldn't be.

@AndreiSavici and @VickieLynch can confirm the problem.

cc: @raquelalvarezbanos @FedeMPouzols @martyngigg as you were also involved in the pull request

AndreiSavici, 2016-08-08:
I tried the following:
```
Load(Filename='CNCS_7860', OutputWorkspace='CNCS_7860_event')
ConvertToMD(InputWorkspace='CNCS_7860_event', QDimensions='Q3D', dEAnalysisMode='Elastic', QConversionScales='HKL', OutputWorkspace='b')
```
Then from the MantidPlot GUI I try to create another MD workspace, with a different name. Sometimes it works, sometimes it crashes. It depends on the name of the output workspace

peterfpeterson, 2016-08-09:
This is **not** a MD specific issue. Another way to produce it:
42. `CreateEmptyTableWorkspace` with any name for the workspace
42. `Add` gui/dialog
42. :boom: with a message about `Attempt to free invalid pointer`

PR #17167 (2016-08-11): Fix segfault when using MD workspaces
**Description of work.**
As part of the changes following #16967 the returned type was changed from an unordered map to a vector. 
In `WorkspaceProperty.h` the loop erases various elements which invalidates the vector iterator which is subsequently dereferenced in the for loop crashing the program.

I've checked all the changed files from the previous PR for any other similar cases however this seems to be the only place where an iterator is invalidated in a loop

All test cases documented in #17165 have been fixed however if there are any more test cases  [... cut]
closed 2016-08-11 by FedeMPouzols