mantidproject/mantid #17184: Set parameters from log: more flexible combinations

tom-perkins, 2016-08-10:
See #16494: introduced option to set parameters to log value in Edit Local Parameters dialog (multi-dataset fitting).

Users would like to have greater flexibility here. For example, instead of setting a parameter to one log value like `Field_Z`, they may want to use a sum like `Field_Main + Field_Z`, or log value plus constant, or a conversion like field -> frequency. 

There should be some way to specify a combination of log values that will be used, perhaps a box to type into or a separate dialog to build the selected combination.

tom-perkins, 2016-08-11:
It would be best to have the ties to log values in the fit function string itself. This would be more general and you'd also be able to change the data and use the same function without having to re-set all the parameter values to logs. See [`ConvertAxisByFormula`](http://docs.mantidproject.org/nightly/algorithms/ConvertAxisByFormula-v1.html) for an example of how this could be done.

tom-perkins, 2016-10-14:
Priority for release 3.9: this has been asked for by scientists and would be useful to a number of users

Anders-Markvardsen, 2017-03-04:
unclear to me to what extent this is a requirement driving by a muon need

closed 2021-03-01 by AnthonyLim23