mantidproject/mantid #17211: Engg:: Create script to process event data to GSS files

DavidFair, 2016-08-12:
As the instrument scientists start moving to running more experiments with event workspaces we need to write a script which processes these workspaces for them

The script should perform the following steps:
Read an Event workspace 
Split the data
Rebin to histogram
Focus the file `(EnggFocus)`
Save a GSS file with params

An example run is as follows
Vanadium: 246043
CeO2: 249918
Run: 250165

Due to this run being extremely large the reading and splitting stage will fail due to an upstream bug in HDF. For this reason the script should be split at this stage so the reading and splitting happens on the patched Mantid build: `https://github.com/mantidproject/mantid/tree/quick_dirty_fix_enginx_large_event_files_christopher_manchester`
A patched executable can be found for testing: `https://www.dropbox.com/s/1kov4m5kkcvm9qe/mantid-3.7.20160705.1624-win64.exe?dl=0`
However as this patch is only applicable in this situation we need the issue to be resolved in the upstream.

FedeMPouzols, 2016-08-12:
The GSAS instrument parameters files are produced by the calibration tab of the Engg GUI, with names like `ENGINX_236516_241391_all_banks.prm` and `ENGINX_236516_241391_bank_North.prm` inside `Calibration`. From there we need the files with parameters for the individual banks.
In addition to those, we need to save the histogram files. These are produced by the focusing tab (the files named like `ENGINX_256663_bank_1.gss` inside `Focus`. 

You can find this in the code in `EnggDiffractionPresenter.cpp` when they use `saveGSS`. The focused files with extension `.gss` are for GSAS, the ones with extension `.dat` are for TOPAS, and the ones with extension `.his` are for OpenGenie.

DavidFair, 2016-09-16:
An initial copy of the script has been sent out - we need feedback on where to go with this.

closed 2017-01-16 by NickDraper