mantidproject/mantid #17250: Engg:: Update EnggCalibrate docs

DavidFair, 2016-08-16:
The documentation for `EnggCalibrate.py` no longer matches the scripts behaviour.
The documentation indicates only two values shall be returned with a 3rd optional param if a certain property is set. 
However 4 values have been returned since #15810 so update documentation to reflect this.

FedeMPouzols, 2016-08-22:
My bad, I must have forgotten to double-check the  doc page when EnggFitDIFCFromPeaks spawned from EnggFitPeaks. Where is the out-of-date text? The doc-test examples look good?
Is it clear what the documentation should say?

DavidFair, 2016-08-22:
I've just looked into it and the documentation for `EnggCalibrate` is fine. I think I was looking at documentation associated with an old release when `diffA` hadn't been implemented hence the confusion.

closed 2016-08-22 by DavidFair