mantidproject/mantid #17270: Engg:: Masking needs to happen at TOF not dSpacing

DavidFair, 2016-08-18:
Currently there are gaps in the data at approx multiples of 10,000 TOF this could be because masking if being applied in TOF and not dSpacing. Reorder the masking so that it happens during the first TOF before dSpacing conversion

DavidFair, 2016-08-19:
Whilst I can replicate these drops in 3.7.2 it appears to be fixed in the latest nightly build. 

- [ ] Verify that it is fixed in the latest builds with the scientists before closing

FedeMPouzols, 2016-08-22:
I think this was fixed here: #16896. If so, this could be marked as "Review required" to close it.

DavidFair, 2016-09-16:
As mentioned I think this has been fixed in #16896 so I'll close this issue as a duplicate

closed 2016-09-16 by DavidFair