mantidproject/mantid #17273: Algorithms with Histogram refactoring - performance improvement candidates

SimonHeybrock, 2016-08-18:
The following Algorithms may have seen performance improvements due to Histogram-rollout related changes. Performance tests should probably be added to track improvements:

Relevant PRs are:

Already merged:
- https://github.com/mantidproject/mantid/pull/16833
- https://github.com/mantidproject/mantid/pull/16958
- https://github.com/mantidproject/mantid/pull/16978
- https://github.com/mantidproject/mantid/pull/17004
- https://github.com/mantidproject/mantid/pull/17038
- https://github.com/mantidproject/mantid/pull/17047

Still open:
- https://github.com/mantidproject/mantid/pull/17121
- https://github.com/mantidproject/mantid/pull/17174

SimonHeybrock, 2016-08-18:
@DimitarTasev A strategy to make gathering the result easier might be:

1. Implement performance tests for all relevant algorithms.
2. Gather results (a few iterations, to get reasonable statistics).
3. Revert **all** changes in the `src/` directory, keeping `test/`. That is, we can do everything in one step. We may need to pay attention to other unrelated changes that we might revert.
4. Gather results (a few iterations, to get reasonable statistics).

This is just a note for now, we should discuss details once we have the list.

DimitarTasev, 2016-08-22:
@SimonHeybrock Do you want me to write the missing performance tests? Also can you suggest a better way to revert files, what I did was check the history of the file and revert to a commit before it was refactored to use HistogramData?

SimonHeybrock, 2016-08-22:
@DimitarTasev Get in touch with @LamarMoore. Did you go through all the linked PRs to see if the list of algorithms above is complete?

You can probably revert all files that were changed in the same PR at once (i.e., revert the whole `src` and `inc` directories to that commit (but keep `test`, if once you added the test).

SimonHeybrock, 2016-08-22:
@DimitarTasev One suggestion:

Work on Algorithms from one pull request at a time.

```bash
# 1. Create branch right before merge commit
git checkout -b performance_test_tmp <commit-right-before-PR-was-merged>
# 2. Implement performance tests, commit to performance_test_tmp
# 3. Run performance tests (store results (xml)).
# 4. Rebase performance_test_tmp onto the merge commit of that PR
git rebase <merge-commit-of-that-PR>
# 5. Run performance tests  (store result (xml))
# 6. Finally: Push and create PR such that new performance tests can be merged into master
```

(I have not tried this, but it should work)

SimonHeybrock, 2016-09-14:
I noticed that also the initial `HistogramData` PR has a few potential algorithm improvements:

ExtractSpectra
https://github.com/mantidproject/mantid/pull/16627/files#diff-26efe394d88bd982e15bc70b4608ae69R336

PolarizationCorrection
https://github.com/mantidproject/mantid/pull/16627/files#diff-8254a0233c9c4aa5c25ec58cd01226caR237

Rebin2D
https://github.com/mantidproject/mantid/pull/16627/files#diff-30415715caad64164828daf988df33a4R208

RebinByPulseTimes
https://github.com/mantidproject/mantid/pull/16627/files#diff-332542ac7846e7718d7a2c41ce6a92b1R63

RebinByTimeAtSample
https://github.com/mantidproject/mantid/pull/16627/files#diff-671e9ff7375bfe7038ea2dd0dab16635R79

ReflectometryTransform
https://github.com/mantidproject/mantid/pull/16627/files#diff-ce6885c348117e37a971e569185ce9c1R195

SimonHeybrock, 2016-09-14:
FindPeaks
https://github.com/mantidproject/mantid/pull/16958/files#diff-d2ea42be211e29aa590882ed62c8774bR643

ScaleX (test added, performance results missing)
StripPeaks(test added, performance results see PR?)

SimonHeybrock, 2016-09-15:
@LamarMoore @DimitarTasev I looked through some changes again. The following two algorithms are listed above, but I think they should show no change: 
- CorrectKiKf
- CorrectToFile

Some more that we overlooked previously:

AbsorptionCorrection
https://github.com/mantidproject/mantid/pull/16833/files#diff-da2f4803dedc2187a4d76c3daefdd171R147

ConvertEmptyToTof
https://github.com/mantidproject/mantid/pull/16833/files#diff-6b604080e9a1d44fe337a6b4c6001fbfR143

ConvertToMatrixWorkspace
https://github.com/mantidproject/mantid/pull/16833/files#diff-9683c0aaa5149109e7672dea36297e15R58

CrossCorrelate
https://github.com/mantidproject/mantid/pull/16833/files#diff-36b29bdd0037bd60ca45e07c7d2208c5R180

ExtractFFTSpectrum
https://github.com/mantidproject/mantid/pull/16833/files#diff-f94fcfdc0b1a84b0cb60b99813c459dcR64

I will have a look at those and let you know if I need help.

DimitarTasev, 2016-09-15:
@SimonHeybrock For the CorrectKiKf and CorrectToFile algorithms, I saw in my notes that they use sharedX now, doesn't that mean they should have about a 1/3 visible improvement on larger workspaces?

SimonHeybrock, 2016-09-15:
@DimitarTasev Not automatically. Previously there was `setX` and `refX` (or `ptrX`). If those methods were used, there is no change. If they were not used, there should be a change. Note that `refY` or `refE` did not exist, so `setSharedY` and `setSharedE` should almost always give an improvement.

PR #17532 (2016-09-19): Release notes v3.8 update: Histogram performance improvements
Added section to release notes that lists algorithms that have seen performance improvements due to refactoring during the `HistogramData` rollout.

Special thanks to @LamarMoore and @DimitarTasev who collected most of the results and implemented new performance tests.

**To test:**

No code changes.
Fixes #17273.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the c [... cut]
closed 2016-09-19 by martyngigg