mantidproject/mantid #17296: Engg GUI: fails to copy output calibration files to "General" directory when the files are not written fast enough

FedeMPouzols, 2016-08-22:
### Expected behavior

Output calibration files should be produced in the `User` output directory and also copied into the `output

### Actual behavior

I could replicate this on debian. I get messages like:
```
Cannot copy the file /home/fedemp/EnginX_Mantid/User/RB0003fff/Calibration/ENGINX_236516_241391_bank_North.prm to the general/all users directories because it cannot be read.
```
But when I go and check the file is there!
You'll get this only the first time, if you repeat the process, because the files are already there, they will be copied without issues. 

The problem is I think that the files are produced in a Python script which runs in a separate thread, immediately after starting that script, we try to copy the file. We'll need to be more patient. This happens in `EnggDiffractionPresenter::doCalib` when they call `copyToGeneral()`.

### Steps to reproduce the behavior

Run a calibration with the GUI, for example using the data / run numbers given in #17277. Note this bug may be elusive, you need to be lucky to catch it.

### Platforms affected

Probably all.

DavidFair, 2016-08-22:
@FedeMPouzols would it be refactoring the code so `copyToGeneral` is called by `writeOutCalibFile` on success? This would solve the race condition and also give us better error checking before attempting to copy the saved calibration file.

FedeMPouzols, 2016-08-22:
That could be one option. I'm not totally sure how to fix the issue without too much complexity. In principle `writeOutCalibFile()` is running the script and getting the status string right before calling `copyToGeneral()`. The output file is created in a `with` estatement which should close the file, but from the close to the sync there seems to be some significant delay. Maybe `os.fsync` or `sync` would be a simple solution.

DavidFair, 2016-08-22:
Does `runPythonCode` spin up another thread to complete this or is the call blocking? If the call isn't blocking we will still get instances where Poco reports that the file doesn't exist in `copyToGeneral( )` before Python has called `fsync`

FedeMPouzols, 2016-08-22:
The python code is run by `PythonRunner::runPythonCode()` which uses a Qt signal that goes to `ApplicationWindow::runPythonScript()`. So yes, an `fsync` is not sufficient I'm afraid.

DavidFair, 2016-09-08:
So as an interim fix I've called `writeOutCalibFiles` twice for each directory. This causes us to do a lot of unnecessary work but it has a better chance that at least one calibration file writes out.

The place I think we need to look is `writeOutCalibFile` which to quote the comment at the top of it:
```
  // TODO: this is horrible and should be changed to avoid running
  // Python code. Update this as soon as we have a more stable way of
  // generating IPARM/PRM files.
```

For a more long term proposal: If we adapted the Python script under EnggUtils to allow it to accept multiple directories and write out to both files simultaneously or copy the data without reprocessing. At least it moves the responsibility of ensuring the files are written out of the presenter and onto the writer?

closed 2016-09-19 by martyngigg