mantidproject/mantid #17299: Engg:: DiffFittingPresenter tests expected call count not matched

DavidFair, 2016-08-22:
The unit tests for `EnggDiffFittingPresenter` expect calls to `getFittingRunNumVec()` which aren't satisfied. A lot of work has recently been completed on the process fitting uses to load and determine filenames so this could be related.

We should investigate if the unit tests need updating to match current behaviour or if the current behaviour is flawed and needs fixing.

FedeMPouzols, 2016-08-22:
Aha, the test is not failing probably because several of the test cases do not do 
```
TSM_ASSERT( 
        "Mock not used as expected. Some EXPECT_CALL conditions were not satisfied",
        testing::Mock::VerifyAndClearExpectations(&mockView))
```
We should ensure that all the test cases do this verification.

DavidFair, 2016-09-29:
Fixed by #17432

closed 2016-09-29 by DavidFair