mantidproject/mantid #17301: Engg GUI fitting tab: refactor

FedeMPouzols, 2016-08-22:
Following on  #17276, the code needs refactoring before adding more functionality/fixes. This concerns:
- Names of variables and methods
- Documenting methods and parameters
- Structure of methods (simplify for example `EnggDiffFittingPresenter::browsedFile()`,
`EnggDiffFittingPresenter::updateFittingDirVec()`, and possibly `EnggDiffFittingPresenter::fittingRunNoChanged()`).

DavidFair, 2016-08-22:
Making a checklist of refactoring locations to complete 
- [ ] `updateFittingDirVec`
- [ ] `browsedFile`
- [ ] `fittingRunNoChanged`

DavidFair, 2016-08-23:
In `enableMultiRun` we currently call `findFilePathsFromBaseName` multiple times. 
This means we walk the directory for every file which can be very expensive.

@FedeMPouzols One possible solution is we could possibly look at getting `findFilePathsFromBaseName` to return a mapping of files onto their found full file paths. This means we could pass in a vector with `n` number files to find in a directory and get their paths in a single call instead of multiple calls?

FedeMPouzols, 2016-08-23:
@DavidFair: that makes perfect sense. We'd scan the directory and produce the map just once at the beginning. I'd definitely put it in the list. Addressing all these issues may take some time and/or unveil other issues, so the list could be split into several GitHub issues/PRs.

DavidFair, 2016-08-23:
Ok i'll make a list as I go and open them in one big batch.
New issue candidates:
- [ ] Return mapping from `findFilePathsFromBaseNames` of base names onto maps to avoid multiple calls

closed 2016-09-21 by martyngigg