mantidproject/mantid #17364: Should SingletonHolder<T>::Instance()  use double-checked locking?

quantumsteve, 2016-08-30:
### Expected behavior

singleton can be called from multiple threads

### Actual behavior

I noticed that [SingletonHolder<T>::Instance()](https://github.com/mantidproject/mantid/blob/adabe9c92d3b45d8125c7d1bb5514552dd1e68b5/Framework/Kernel/inc/MantidKernel/SingletonHolder.h#L79) just has a singe check on `pinstance`. Since it is called concurrently from multiple threads, should this be replaced with  [double-checked locking](https://en.wikipedia.org/wiki/Double-checked_locking), `std::call_once`, or (ideally) a static initializer?

### Steps to reproduce the behavior

### Platforms affected

All

quantumsteve, 2016-08-30:
@martyngigg @peterfpeterson

martyngigg, 2016-08-31:
There certainly should be some locking - I assumed there was.

`std::call_once` would seem a nice fit here - it does what we need and has a good name for this circumstance.

peterfpeterson, 2016-08-31:
I'm not the right person to comment on this.

PR #17402 (2016-09-06): Use std::call_once to ensure pInstance isn't initialized multiple times
Description of work.

This changes `SingletonHolder<T>::Instance` to use[ `std::call_once`](http://www.cplusplus.com/reference/mutex/call_once/). The current implementation with a single check is not necessarily thread-safe.

**To test:**

Thorough code review. This function is a core piece of the Mantid Framework.

Fixes #17364.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable [... cut]
closed 2016-09-06 by peterfpeterson