mantidproject/mantid #17384: Memory not released when deleteing workspace open in spectrum viewer

DavidFair, 2016-09-01:
### Expected behavior
When a workspace is open in the spectrum viewer and is deleted in Mantid the memory should be released.
### Actual behavior
The window for the spectrum viewer is closed and the workspace is removed from the ADS however the memory is not released.
### Steps to reproduce the behavior
Open a large workspace so the additional memory usage is visible for example under system test data:
`PG3_2538_event.nxs` (or rebin a workspace with an extremely small bin width).

Open the spectrum viewer for that workspace
Without closing the spectrum viewer delete the workspace
Observe that the memory is not released but the workspace is deleted. 

### Platforms affected
Windows - Probably All

mantid-roman, 2016-09-01:
The simplest solution is to set to `true` the `Qt::WA_DeleteOnClose` attribute of the viewer window in `MantidUI::showSpectrumViewer()` method (line 803 of MantidUI.cpp).

DavidFair, 2016-09-01:
This resolves the issue - thanks for the help. I'll open a PR with the changes

PR #17386 (2016-09-02): Fix spectrum viewer not releasing memory on workspace deletion
Description of work.
If users had the spectrum viewer open and deleted a workspace the memory used by that workspace was not released. This change ensures the memory is correctly deallocated when the workspace is deleted

**To test:**
Open a large workspace so the additional memory usage is visible for example under system test data:
`PG3_2538_event.nxs` (or rebin a workspace with an extremely small bin width).

Open the spectrum viewer for that workspace
Without closing the spectrum viewer delete the workspace

If the memory usage [... cut]
closed 2016-09-02 by peterfpeterson