mantidproject/mantid #17390: Move monitor integration to `ReflectometryWorkflowBase`

raquelalvarezbanos, 2016-09-01:
In `CreateTransmissionWorkspace.cpp`, [line 168](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/CreateTransmissionWorkspace.cpp#L168):

```
// Monitor integration ... can this happen inside the toLam routine?
```

I think the answer is yes. If so, monitor integration should be moved to `ReflectometryWorkflowBase::toLamMonitor()`

PranavBahuguna, 2016-09-02:
While it does seem possible to migrate monitor integration to `ReflectometryWorkflowBase::toLamMonitor()`, I noticed that in `ReflectometryWorkflowBase::toLam()`, the monitor integration code would not be carried out if `monitorIndex` or `backgroundMinMax` were not initialised, in which case `makeUnityWorkspace()` would be executed instead of `toLamMonitor()`. Does this matter?

In addition, I could avoid this problem by placing the monitor integration code in `toLam()` itself rather than `toLamMonitor()`.

raquelalvarezbanos, 2017-02-06:
No longer valid, we now have a new version of `CreateTransmissionWorkspace`.

closed 2017-02-06 by raquelalvarezbanos