mantidproject/mantid #17405: Refactor CompareSampleLogs

mganeva, 2016-09-05:
### Expected behavior
Experience shows, that sometimes numeric sample logs are represented as strings. However, users expect them to be compared as numbers (with certain tolerance).

The aim of the refactoring is to implement the behaviour expected by users, e.g. compare temperatures as numbers, even if they have type 'String'.

### Platforms affected
All

Please, assign me.

FedeMPouzols, 2016-09-05:
@mganeva: unfortunately I think there's no support for assigning GitHub issues to "external" developers (not in the repository list). It seems that you'll be able to assign yourself only in your own fork.

mganeva, 2016-09-05:
@FedeMPouzols: no problem. I just wanted to communicate, that I am working on it right now and there is no need to assign someone else to this issue. I hope, it is okay.

soininen, 2016-09-05:
What about adding the 'State: In Progress' label?

mganeva, 2016-09-05:
@soininen: I've submitted the pull request already and I am waiting for review. I think, that the label 'In Progress' is for that pull requests which should not be reviewed yet.

PR #17406 (2016-09-12): Refs #17405. Refactoring of CompareSampleLogs.
Refactoring of ComateSampleLogs algorithm to fix the mentioned issue and some minor code improvements. 

2 new unit tests.

**To test:**
Check that nothing has got broken. The simple test for new behavoiur:
```python
ws1 = CreateSampleWorkspace()
ws2 = CreateSampleWorkspace()
AddSampleLog(ws1, 'chopper_speed', '13995', 'String')
AddSampleLog(ws2, 'chopper_speed', '14004', 'Number')
result = CompareSampleLogs([ws1, ws2], 'chopper_speed', 10.0)
print result
```
the result must be an empty string.

Fixes #17405. 

Does not need to be in the release n [... cut]
closed 2016-09-12 by peterfpeterson