mantidproject/mantid #17427: Improve `OFFSPECReflRedOneAutoPolarizationCorrection` systemtest

raquelalvarezbanos, 2016-09-08:
The systemtest looks like this:

```
inputWorkspace = Load("OFFSPEC00033767.nxs")
transmissionGroup = Load("OFFSPEC00033772.nxs")
#set up our transmission workspace
transWorkspace = CreateTransmissionWorkspaceAuto(transmissionGroup,
        AnalysisMode="MultiDetectorAnalysis",
        ProcessingInstructions="110-120",
        WavelengthMin=2.0,
        WavelengthStep=0.05,
        WavelengthMax=12.0)
# set up our efficiency constants
CRho=[1]
CAlpha=[1]
CAp=[1]
CPp=[1]
#run reflectometryReductionOneAuto
__, _IvsLam_polCorr,__ = ReflectometryReductionOneAuto(inputWorkspace, AnalysisMode="MultiDetectorAnalysis",
        ProcessingInstructions="110-120",
        FirstTransmissionRun=transWorkspace,
        ThetaIn="1.2",WavelengthMin=2.0,
        WavelengthStep=0.05,
        WavelengthMax=12.0,CorrectionAlgorithm='None',
        PolarizationAnalysis='PA', MomentumTransferStep=0.1,
        CPp=CPp,CAp=CAp,CRho=CRho,CAlpha=CAlpha)
```

Polarization constants are all set to `1`, perhaps a more clever test would use different constants for each workspace?

I referred to `PolarizationCorrection` documentation (which is the algorithm applying the polarization correction) to see how those constants are used but there are no user examples. At least one example (ideally a sensible one) should be added.

gemmaguest, 2018-03-21:
We should check whether this is still relevant / useful

gemmaguest, 2020-07-20:
Closing this because we will add new system tests as and when examples are provided from scientists

closed 2020-07-20 by gemmaguest