mantidproject/mantid #17430: Improve unit tests for `ReflectometryReductionOne`

raquelalvarezbanos, 2016-09-08:
See https://github.com/mantidproject/mantid/issues/17429

- All the input properties should be covered by unit tests. We can create our own workspaces, I don't think we need to use real data.

PranavBahuguna, 2016-09-14:
@raquelalvarezbanos I have been attempting to write tests for `CreateTransmissionWorkspaceTest::test_detector_position_correction_multi_detector()`. I found that while the first detector's position had changed, the other 3 detectors did not. After analysing `SpecularReflectionPositionCorrect`, I believe that algorithm completely lacks any code whatsoever for dealing with multiple detectors, and thus this seems to be quite a serious issue.

raquelalvarezbanos, 2016-10-11:
@PranavBahuguna is there a PR for this?

PranavBahuguna, 2016-10-11:
@raquelalvarezbanos Apparently not, should I create one?

raquelalvarezbanos, 2016-10-11:
If it is ready for testing, yes please, create one.

raquelalvarezbanos, 2016-10-21:
@PranavBahuguna Did you create a PR for this?

gemmaguest, 2018-03-21:
We should check whether this is still relevant / useful

gemmaguest, 2020-08-06:
Closing as stale because we are unlikely to get around to this

closed 2020-08-06 by gemmaguest