mantidproject/mantid #17455: VSI Crashes when swapping between views with two sources

samueljackson92, 2016-09-09:
I've found that I can make VSI crash by opening two MDHisto workspaces in VSI in the standard view and swapping to the Multislice view in the current master. I tried the same thing on v3.6.0 (what I happened to have on my machine) and it didn't crash.

### Expected behavior
Mantid should not crash with multiple sources and changing view.

### Actual behavior
Mantid crashes when using multiple sources and changing view.

### Steps to reproduce the behavior
I'm creating some dummy MDWorkspaces using the following scripts:

```python
from mantid.api import Projection

to_cut = CreateMDWorkspace(Dimensions=4, Extents=[-1,1,-1,1,-1,1,-10,10], Names="H,K,L,E", Units="U,U,U,V")
# Add two fake peaks so that we can see the effect of the basis transformation
FakeMDEventData(InputWorkspace='to_cut', PeakParams=[10000,-0.5,0,0,0,0.1])
FakeMDEventData(InputWorkspace='to_cut', PeakParams=[10000,0.5,0,0,0,0.1])

#Good slicing
projection = Projection([1,0,0], [0.5,-1,0], [0, 0, 1])
proj_ws = projection.createWorkspace()
#out_md = CutMD(to_cut, Projection=proj_ws, PBins=([0.1], [0.1], [0.1], [-5,5]), NoPix=True)
BinMD(InputWorkspace='to_cut', AxisAligned=False, 
BasisVector0='A, r, 1, 0, 0, 0', 
BasisVector1='B, r, 0.5, 1, 0, 0', 
BasisVector2='C, r, 0, 0, 1, 0', 
BasisVector3='E, V, 0, 0, 0, 1', 
OutputExtents='-1,1,-1,1,-1,1,-5,5', OutputBins='200,300,200,1', NormalizeBasisVectors=False, 
OutputWorkspace='slicedA')

#Bad slicing
projection = Projection([1,0,0], [-0.5,1,0], [0, 0, 1])
proj_ws = projection.createWorkspace()
#out_md2 = CutMD(to_cut, Projection=proj_ws, PBins=([0.1], [0.1], [0.1], [-5,5]), NoPix=True)
BinMD(InputWorkspace='to_cut', AxisAligned=False, 
BasisVector0='A, r, 1, 0, 0, 0', 
BasisVector1='B, r, -0.5, 1, 0, 0', 
BasisVector2='C, r, 0, 0, 1, 0', 
BasisVector3='E, V, 0, 0, 0, 1', 
OutputExtents='-1,1,-1,1,-1,1,-5,5', OutputBins='200,300,200,1', NormalizeBasisVectors=False, 
OutputWorkspace='slicedB')
```

### Platforms affected
Saw it on OSX, but could well be on others.

samueljackson92, 2016-09-12:
@quantumsteve @OwenArnold @AntonPiccardoSelg

NickDraper, 2016-10-03:
This is not common practice and has not been reported by instrument scientists, so shouldn't hold up r3.8, moved to r3.9

samueljackson92, 2016-10-21:
As discussed, we realised that this is not commonly done. Hence remove the high priority status.

samueljackson92, 2017-10-05:
This appears to have since been fixed.

closed 2017-10-05 by samueljackson92