mantidproject/mantid #17479: Load not loading VanCurvesWorkspace

DimitarTasev, 2016-09-13:
### Expected behavior

The `VanCurvesWorkspace` is properly loaded into Mantid after selecting it through Load.

### Actual behavior

Nothing is loaded and the following error is produced in the Log:

```
Error in execution of algorithm LoadNexusProcessed:
Attempt to load from an empty dataset /mantid_workspace_1/workspace/axis2
Error in execution of algorithm Load:
Attempt to load from an empty dataset /mantid_workspace_1/workspace/axis2
```

### Steps to reproduce the behavior

Try to load a VanCurvesWorkspace into mantid

### Platforms affected

All

DavidFair, 2016-09-14:
It looks like the issue is axis2 is not being saved out by `NexusFileIO` - specifically `sAxis` is a text type so has no units and at the check `  if (!sAxis->isText())` trys to access the data via `sAxis->label(i)` which holds nothing hence nothing is saved out.

DavidFair, 2016-10-21:
I've talked to @DimitarTasev and he is happy to look into this in the mean time.

PR #17886 (2016-11-03): LoadNexusProcessed can now handle the special case of TextAxis with empty label
### Description of work.

`LoadNexusProcessed` will now properly handle a TextAxis with an empty label.
2 Unit tests have also been added to test for loading a TextAxis with an empty label, and for a TextAxis with a non-empty label

**To test:**
- Run LoadNexusProcessed unit tests
- Try to load a workspace with a TextAxis and empty label ([van_corrections ws](https://drive.google.com/file/d/0B2Mj8e56Z8xTWUE2eGdMRDBjQ1k/view?usp=sharing)) 

Fixes #17479 .

_Does not need to be in the release notes._
---
#### Reviewe [... cut]
closed 2016-11-03 by martyngigg