mantidproject/mantid #17509: Clean up FABADA minimizer code

KarlPalmen, 2016-09-15:
Improve to code of FABADA minimizer to ensure it of acceptable quality, conforms to coding standards and is well structured with small focused classes/methods/functions.

Work on this is best done after all urgent issues arising out of beta testing have be resolved.

martyngigg, 2016-09-15:
Please make sure this includes things such as naming methods and variables in `camelCase` style for consistency with the rest of the code.

Also, in `FabadaMinimizer.cpp` there are several constants defined, such as `convergenceMaxIterations`. These should be capitalised with underscores: `CONVERGENGE_MAX_ITERATIONS` to distinguish them from local variable names.

raquelalvarezbanos, 2016-09-15:
I'll take care of this issue. I noticed the style was not following standards, but given the proximity of the code freeze I needed to create a PR. I will fix all the issues with style during the maintenance period.

raquelalvarezbanos, 2016-10-14:
Maintenance task. There's an open PR pending testing. I've been in contact with Bartoumeu and should be testing this soon. Therefore marking against release 3.9.

PR #20020 (2017-10-12): Clean up FABADA minimizer code
Retakes leftover pull request, solving merge conflicts: [(https://github.com/mantidproject/mantid/pull/17671)]

It was pointed out that FABADA minimizer code should:
- Use camelCase
- Be structured in small focused methods
- Constants should be capitalised with underscores

Fixes [#17509](https://github.com/mantidproject/mantid/issues/17509)

Since it has been not updated for a long time, the merge step, in order to avoid merge conflicts, has introduced a lot of commits.

**To test:**

Fixes #17509 . 

**Release Notes** 
Does not need to be in the release notes.
 [... cut]
closed 2017-10-12 by martyngigg