mantidproject/mantid #17513: Logging makes execution 3x slower on Windows.

quantumsteve, 2016-09-15:
### Expected behavior

Logging should (at least by default) have a minimal effect on the execution time. 

### Actual behavior

System Test: HFIRTestsAPIv2.HFIRTestsAPIv2
time taken (defaults)
282.54 second
284.69 seconds
289.60 seconds

time taken (disable logging by removing code in ThreadSafeLogStreamBuf::writeToDevice)
92.00 seconds
91.25 seconds
96.05 seconds

I would expect other system tests to show similar results.

### Steps to reproduce the behavior

From the build directory run `systemtest.bat -C Release -R HFIRTestsAPIv2.HFIRTestsAPIv2` 

Comment out [ThreadSafeLogStream.cpp L31-L39](
(https://github.com/mantidproject/mantid/blob/e84e49c512eca65625f4add4db1beeb0b0127c0e/Framework/Kernel/src/ThreadSafeLogStream.cpp#L31)

rebuild and rerun the system test HFIRTestsAPIv2.HFIRTestsAPIv2 .

### Platforms affected

I can only reproduce this issue on Windows

I tried doing the same timings on OS X and the results were all within 3 seconds.

quantumsteve, 2016-09-15:
@martyngigg Is this a known issue? Please reassign as appropriate.

quantumsteve, 2016-09-15:
`systemtest.bat -C Release -R MERLINReduction`

(default)
101.57 s
96.13 s
98.93 s

(disable logging by removing code in ThreadSafeLogStreamBuf::writeToDevice)
64.20 s
68.44 s
68.74 s

martyngigg, 2016-09-16:
@quantumsteve I'm not sure this have been timed before so I can't say if this is new behaviour or not.

I wonder if the locks are more expensive to obtain on Windows?

quantumsteve, 2016-09-16:
@martyngigg The lock on line 37 doesn't appear to be causing this issue. I commented out lines 37 & 38 and reran `systemtest.bat -C Release -R MERLINReduction`

96.55 sec
105.42 sec
107.77 sec

The elapsed time isn't any better than master.

martyngigg, 2016-09-16:
@quantumsteve That's interesting. Is it all logging or are the results the same if you just disable file logging?

martyngigg, 2016-09-16:
@quantumsteve I was just browsing the Poco code and I notice in [FileChannel](https://github.com/pocoproject/poco/blob/develop/Foundation/src/FileChannel.cpp#L47) that it flushes every time a log message is written. It looks like there is a `flush` property that can control this.

Could you try setting `logging.channels.fileChannel.flush = false` in `~/.mantid/Mantid.user.properties` and see if that makes a difference?

quantumsteve, 2016-09-16:
@martyngigg Yes, the `flush` property appears to be the issue. After adding the line above to my `Mantid.user.properties` and forcing `systemtest` to use that file, the results are faster than either result above.

systemtest.bat -C Release -R MERLINReduction --disablepropmake
65.25 sec 
59.90 sec
63.35 sec

systemtest.bat -C Release -R HFIRTestsAPIv2.HFIRTestsAPIv2 --disablepropmake
62.97 sec
63.36 sec
65.57 sec

martyngigg, 2016-09-16:
@quantumsteve That's an easy fix then. Presumably now the OS will just handle flushing the buffers when the are full?

I'll make a PR for the change.

quantumsteve, 2016-09-16:
Is there a bug in the Windows implementation of `LogFile`? 

On Linux and OSX, the difference is [whether or not `\n` or `std::endl` is used]( ]https://github.com/pocoproject/poco/blob/e87a8fe962ec3424b73305117343a167afe79b9a/Foundation/src/LogFile_STD.cpp#L43)

On Windows, flush adds a [call to FlushFileBuffers](https://github.com/pocoproject/poco/blob/e87a8fe962ec3424b73305117343a167afe79b9a/Foundation/src/LogFile_WIN32.cpp#L53). 

[This stackoverflow post](http://stackoverflow.com/questions/14290337/is-fwrite-faster-than-writefile-in-windows) suggests the later is MUCH slower.

martyngigg, 2016-09-17:
Hmm, that's interesting. I wonder why they don't just use the standard implementation on Windows too?

Is it worth raising an issue for [pocoproject/poco](https://github.com/pocoproject/poco)?

quantumsteve, 2016-09-17:
If `LogFile_STD` or a similar change to `LogFile_WIN32` addresses the performance regression on Windows, it would be nice to have this also addressed upstream. What is the best way to generate a small test file reproducing the issue?

PR #17531 (2016-09-19): Disable log file flushing by default
The Poco `FileChannel` requests that `LogFile` buffers are flushed on receipt of each newline character. These changes set the `.flush` property to `false` to leave this decision up to the OS.

**To test:**

See instructions in the issue to reproduce the issue.

Fixes #17513

[Release notes updated](https://github.com/mantidproject/mantid/commit/6d1bc6fe16453aea8c02708600acc2a64f9b14c2)

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is th [... cut]
closed 2016-09-19 by SimonHeybrock