mantidproject/mantid #17536: Mantid crashes when plotting spectrum that is out of range

raquelalvarezbanos, 2016-09-20:
During unscripted testing of the Reflectometry interface (Windows, Mantid version: 3.7.20160916.1815) I found the following bug (unrelated to Reflectometry):

Run the following script:

```
inter1 = Load(r'D:\mantid-tests\mantid-build\ExternalData\Testing\Data\SystemTest\INTER00013460.nxs')
inter2 = Load(r'D:\mantid-tests\mantid-build\ExternalData\Testing\Data\SystemTest\INTER00013462.nxs')
```

Right-click on `inter1` and plot one of the spectra, for instance number 2. Drag and drop `inter2` to the plot and select a spectrum out of the range, for instance spectrum number 6. The red star will be shown with a message saying that the spectrum is not in the range available. If you click on `Canclel` or `Plot all` Mantid crashes. Even if you correct the spectrum number to a spectrum that does exist (for instance spectrum number 5) and click on `OK`, Mantid crashes. It also crashes if you close the dialog, so it seems that if you enter a wrong spectrum number by mistake, one way or another Mantid will crash.

raquelalvarezbanos, 2016-09-20:
I am leaving this unassigned as today I am in a course and won't be able to have a look at this

reimundILL, 2016-09-20:
It may help: on Ubuntu 14.04, mantidnightly 20160916, I can load and plot all spectra 1-5 as expected in various order. (Thus, cannot reproduce the error).

reimundILL, 2016-09-20:
I mean, I tested plotting invalid spectra 0, 6, 7 etc as well in between of valid spectrum numbers. I noticed to count the spectra from 1 to 5 instead of 0 to 4.

tom-perkins, 2016-09-20:
I can reproduce the error on Windows 7. Here is the call stack:
[image]
(There is no "Waterfall plot" option displayed in the dialog)

tom-perkins, 2016-09-20:
I think the answer is to initialise the pointers `m_waterfallOpt` and `m_tiledOpt` to null in the constructor - this fixes the error for me.

closed 2016-09-20 by FedeMPouzols

raquelalvarezbanos, 2016-09-20:
Thanks for fixing @tom-perkins !

PR #17545 (2016-09-20): Fix crash plotting invalid spectrum number
Fix crash seen when plotting an invalid spectrum number.
(Two pointers were uninitialized)

**To test:**

Follow the example in the issue and verify there is no crash.

Fixes #17536.

*No release notes* (bug not present in previous release)

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it we [... cut]