mantidproject/mantid #17551: Renaming workspaces in a group allows two workspaces to share a single name

DavidFair, 2016-09-20:
### Expected behavior
On renaming a workspace the `Override Existing` option should mean if you rename a workspace to an existing name in a group that is overridden
### Actual behavior
Two workspaces share a single name and Mantid enters an undefined state.
### Steps to reproduce the behavior
Run this script 
```python 
CreateSampleWorkspace(OutputWorkspace='bob', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob1', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob2', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob3', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
GroupWorkspaces(InputWorkspaces='bob,bob1,bob2,bob3', OutputWorkspace='NewGroup')
RenameWorkspace(InputWorkspace='bob1', OutputWorkspace='bob')
```
If two workspaces called `Bob` exist in the group this bug is present
### Platforms affected
All

DavidFair, 2016-09-20:
The ADS handles this correctly - it appears that the issue lie with the group updating to remove one of the references to the workspace

DavidFair, 2016-09-21:
```python
CreateSampleWorkspace(OutputWorkspace='bob', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob1', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob2', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob3', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
group = GroupWorkspaces(InputWorkspaces='bob,bob1,bob2,bob3', OutputWorkspace='NewGroup')

nameList = group.getNames()
print nameList

RenameWorkspace(InputWorkspace='bob1', OutputWorkspace='bob')

# Update the handle to the group
group = mtd["NewGroup"]
nameList = group.getNames()
print nameList
```

This currently prints two `Bob` workspaces in the group indicating the presence of this bug

martyngigg, 2016-09-21:
⁠⁠⁠⁠AnalysisDataService.getObjectNames()﻿⁠⁠⁠⁠ looks correct but ﻿⁠⁠⁠⁠WorkspaceGroup﻿⁠⁠⁠⁠ hasn't received a delete signal that the old one has been removed for some reason.

Can you check if this bug also existed in `3.7.2` & move this back to `3.9` if so?

DavidFair, 2016-09-21:
This bug exist in 3.7.2 so I'll move this to 3.9 release

closed 2016-10-26 by AndreiSavici