mantidproject/mantid #17559: New WorkflowAlgorithm IndirectILLFixedWindowScans for IN16B

reimundILL, 2016-09-21:
### Expected behavior

The aim is to implement a workflow algorithm for fixed window scans of the ILL IN16B instrument. Currently, only the workflow algorithm for quasi elastic scans QENS (https://github.com/mantidproject/mantid/issues/16753) is implemented. Inelastic and elastic fixed-window scans have Nexus file entries, where the velocity is 1 and energy is 0 (inelastic) or unequal zero (elastic).

Previous work is done by Spencer Howells for ILLElasticScans.

Make use of https://github.com/mantidproject/documents/blob/master/Project-Management/ILL/WorkspaceAlgorithmRunner.md and SelectNexusFilesByMetadata (https://github.com/mantidproject/mantid/pull/17361)

### Platforms affected

All

SpencerHowells, 2016-09-21:
I've already done some algos
[IndirectILLElasticScan.zip](https://github.com/mantidproject/mantid/files/484824/IndirectILLElasticScan.zip)

reimundILL, 2016-10-25:
Proceed work on branch https://github.com/mantidproject/mantid/tree/16753_newIndirectILLReduction

PR #18131 (2017-01-19): ILL :: Indirect Reduction Framework
This PR implements the reduction framework for IN16B indirect geometry instrument at ILL.
It includes a new version for the loader (which treats single detectors slightly different), 3 new workflow algorithms (deprecating the existing 2) and corresponding upgrade of the GUI. 

The results of the reduction are preliminarily validated against the existing software at ILL (LAMP), and science validation is being run in parallel.

It turned out to be a rather large ticket, but the different parts are in tight relation and have been developed together, so we were  [... cut]
closed 2017-01-19 by SimonHeybrock