mantidproject/mantid #17597: fit function to model QENS data with Teixeira water

jmborr, 2016-09-24:
A fitfunction to model the Q-dependence of the FWHM from a lorentzian fit is already implemented in fit function TeixeiraWater. This fit function is the Lorentzian with its FWHM given by the Teixeira model, in order to directly fit QENS data.

SpencerHowells, 2016-09-25:
I have a Bayes version of this, but it's in Fortran and I'm having problems getting it to work! This would be a big improvement and we could also use the Fabada Bayes fitting package.

SpencerHowells, 2016-09-26:
Can this read the Q values from the workspace to do a sequential fit?

jmborr, 2016-09-26:
@SpencerHowells You mean something like the `WorkspaceIndex` property that was included in [InelasticDiffSphere](https://github.com/mantidproject/mantid/blob/master/Framework/CurveFitting/src/Functions/InelasticDiffSphere.cpp) ?

SpencerHowells, 2016-09-26:
Yes I think so - it's what Dan did.

jmborr, 2016-09-26:
@SpencerHowells I've sent a question regarding this to Slack because I don't understand why there should exists an attribute named `WorkspaceIndex` without a corresponding attribute `WorkspaceName`.

SpencerHowells, 2016-09-27:
For sequential fit we use the algo PlotPeakByLogValue which has a property PassWSIndexToFunction

jmborr, 2016-09-27:
@SpencerHowells  Ah, that is the answer I was looking for. Thanks a lot.

PR #17606 (2016-10-05): fit function to model QENS data with Teixeira water
Fit function to model QENS data assuming a particle undergoing jump-diffusion.

**To test:**

- Review  the code.
- In a MantidPlot session, run the example included in the document page.
- After you run the example, use workspaces "data" and "resolution" to do a manual fit using the "Fit Function" tool of MantidPlot. You may want to import the following string as the model:
`(composite=Convolution,FixResolution=true,NumDeriv=true;
    name=TabulatedFunction,Workspace=resolution,WorkspaceIndex=0,Scaling=1,Shift=0,XScaling=1;
    (name=DeltaFu [... cut]
closed 2016-10-05 by AndreiSavici