mantidproject/mantid #17614: AlignComponents not registering correctly

DavidFair, 2016-09-27:
Align components is not correctly registering hence the documentation page for it and the links to it are not working.

FedeMPouzols, 2016-09-27:
Is this on rhel6? Then it's a known limitation. Some required package from scipy is missing on rhel6.

rosswhitfield, 2016-09-27:
It's registered everywhere except on RHEL6 because `scipy.optimize.minimize` doesn't exist on the version of scipy on RHEL6. We are planning to move the documentation build to RHEL7 after which AlignComponents will appear.

DavidFair, 2016-09-27:
Ahhh I see - I'll close this issue as I knew it wasn't registering hence there are no links to it in the nightly documentation but hadn't looked into the reason why.

closed 2016-09-27 by DavidFair