mantidproject/mantid #17641: Umrella: HistogramData rollout during maintenance period

SimonHeybrock, 2016-09-30:
This is an umbrella issue for HistogramData rollout, see https://github.com/mantidproject/documents/blob/master/Project-Management/TechnicalSteeringCommittee/reports/MaintenanceTasks.md#maintenance-tasks-for-39.

Please pick a chunk of 5 to 10 algorithm at a time:

1. Pick algorithm from the **top**.
2. Write a comment on this issue, listing the algorithms you picked.
3. Make sure no one picked the same algorithm while writing the comment (refresh page).
4. Create an issue, start refactoring, and create a PR.
5. Do not hesitate to create the PR early for an intermediate review, or contact Simon Heybrock or Lamar Moore on Slack when in doubt or when anything is unclear.

Algorithms that use `(data|read)(X|Y|E)`:

I am listing the `cpp` files here for convenience, but you are also responsible for the corresponding unit tests (and header files):

```
Framework/Algorithms/src/PDCalibration.cpp
Framework/Algorithms/src/PDFFourierTransform.cpp
Framework/Algorithms/src/PhaseQuadMuon.cpp
Framework/Algorithms/src/PlotAsymmetryByLogValue.cpp
Framework/Algorithms/src/PointByPointVCorrection.cpp
Framework/Algorithms/src/PolarizationCorrection.cpp
Framework/Algorithms/src/Q1D2.cpp
Framework/Algorithms/src/Q1DWeighted.cpp
Framework/Algorithms/src/Qhelper.cpp
Framework/Algorithms/src/Qxy.cpp
Framework/Algorithms/src/RadiusSum.cpp
Framework/Algorithms/src/RayTracerTester.cpp
Framework/Algorithms/src/ReadGroupsFromFile.cpp
Framework/Algorithms/src/RealFFT.cpp
Framework/Algorithms/src/Rebin2D.cpp
Framework/Algorithms/src/RebinByPulseTimes.cpp
Framework/Algorithms/src/RebinByTimeAtSample.cpp
Framework/Algorithms/src/Rebin.cpp
Framework/Algorithms/src/RebinToWorkspace.cpp
Framework/Algorithms/src/Rebunch.cpp
Framework/Algorithms/src/ReflectometryReductionOne.cpp
Framework/Algorithms/src/ReflectometryWorkflowBase.cpp
Framework/Algorithms/src/Regroup.cpp
Framework/Algorithms/src/RemoveBackground.cpp
Framework/Algorithms/src/RemoveBins.cpp
Framework/Algorithms/src/RemoveExpDecay.cpp
Framework/Algorithms/src/RemoveLowResTOF.cpp
Framework/Algorithms/src/RemoveMaskedSpectra.cpp
Framework/Algorithms/src/ResampleX.cpp
Framework/Algorithms/src/ResetNegatives.cpp
Framework/Algorithms/src/RingProfile.cpp
Framework/Algorithms/src/RRFMuon.cpp
Framework/Algorithms/src/SampleCorrections/MayersSampleCorrection.cpp
Framework/Algorithms/src/SANSDirectBeamScaling.cpp
Framework/Algorithms/src/SetUncertainties.cpp
Framework/Algorithms/src/SmoothData.cpp
Framework/Algorithms/src/SmoothNeighbours.cpp
Framework/Algorithms/src/SofQWCentre.cpp
Framework/Algorithms/src/SofQWNormalisedPolygon.cpp
Framework/Algorithms/src/SofQWPolygon.cpp
Framework/Algorithms/src/SolidAngle.cpp
Framework/Algorithms/src/Stitch1D.cpp
Framework/Algorithms/src/StripVanadiumPeaks.cpp
Framework/Algorithms/src/SumEventsByLogValue.cpp
Framework/Algorithms/src/SumRowColumn.cpp
Framework/Algorithms/src/SumSpectra.cpp
Framework/Algorithms/src/TOFSANSResolutionByPixel.cpp
Framework/Algorithms/src/TOFSANSResolution.cpp
Framework/Algorithms/src/Transpose.cpp
Framework/Algorithms/src/UnaryOperation.cpp
Framework/Algorithms/src/UnwrapMonitor.cpp
Framework/Algorithms/src/UnwrapSNS.cpp
Framework/Algorithms/src/VesuvioL1ThetaResolution.cpp
Framework/Algorithms/src/WeightedMeanOfWorkspace.cpp
Framework/Algorithms/src/WienerSmooth.cpp
[... 208 lines cut ...]
MantidQt/CustomInterfaces/src/Muon/ALCPeakFittingPresenter.cpp
MantidQt/CustomInterfaces/src/Muon/ALCPeakFittingView.cpp
MantidQt/CustomInterfaces/src/Muon/MuonAnalysis.cpp
MantidQt/CustomInterfaces/src/Muon/MuonAnalysisFitDataPresenter.cpp
MantidQt/CustomInterfaces/src/Muon/MuonAnalysisHelper.cpp
MantidQt/CustomInterfaces/src/SampleTransmission.cpp
MantidQt/CustomInterfaces/src/SANSPlotSpecial.cpp
MantidQt/CustomInterfaces/src/SANSRunWindow.cpp
MantidQt/CustomInterfaces/src/Tomography/ImageROIViewQtWidget.cpp
MantidQt/CustomInterfaces/src/Tomography/ImggFormatsConvertViewQtWidget.cpp
MantidQt/CustomInterfaces/src/Tomography/TomographyIfaceViewQtGUI.cpp
MantidQt/CustomInterfaces/test/ALCBaselineModellingModelTest.h
MantidQt/CustomInterfaces/test/ALCBaselineModellingPresenterTest.h
MantidQt/CustomInterfaces/test/ALCPeakFittingModelTest.h
MantidQt/CustomInterfaces/test/MuonAnalysisDataLoaderTest.h
MantidQt/CustomInterfaces/test/MuonAnalysisHelperTest.h
MantidQt/MantidWidgets/src/FitPropertyBrowser.cpp
MantidQt/MantidWidgets/src/InstrumentView/InstrumentActor.cpp
MantidQt/MantidWidgets/src/InstrumentView/InstrumentWidgetPickTab.cpp
MantidQt/MantidWidgets/src/MWView.cpp
MantidQt/MantidWidgets/src/PreviewPlot.cpp
MantidQt/MantidWidgets/src/PropertyHandler.cpp
MantidQt/RefDetectorViewer/src/RefMatrixWSImageView.cpp
Vates/VatesAPI/test/vtkDataSetToScaledDataSetTest.h
```

PranavBahuguna, 2016-09-30:
Picked the following algorithms for #17644:
- Framework/Algorithms/src/PDCalibration.cpp
- Framework/Algorithms/src/PDFFourierTransform.cpp
- Framework/Algorithms/src/PhaseQuadMuon.cpp
- Framework/Algorithms/src/PlotAsymmetryByLogValue.cpp
- Framework/Algorithms/src/PointByPointVCorrection.cpp

LamarMoore, 2016-09-30:
@PranavBahuguna  excellent you have already started. When your PR is ready can you assign either myself or @SimonHeybrock  to review.

samueljackson92, 2016-09-30:
Picked the following algorithms for #:

 - Framework/Algorithms/src/PolarizationCorrection.cpp
 - Framework/Algorithms/src/Q1D2.cpp
 - Framework/Algorithms/src/Q1DWeighted.cpp
 - Framework/Algorithms/src/Qhelper.cpp
 - Framework/Algorithms/src/Qxy.cpp

DimitarTasev, 2016-10-03:
Picked the following algorithms for #17661:
- Framework/Algorithms/src/RadiusSum.cpp
- Framework/Algorithms/src/RayTracerTester.cpp
- Framework/Algorithms/src/ReadGroupsFromFile.cpp
- Framework/Algorithms/src/RealFFT.cpp
- Framework/Algorithms/src/Rebin2D.cpp

LamarMoore, 2016-10-03:
Picked the following algorithms:
Framework/Algorithms/src/RebinByPulseTimes.cpp
Framework/Algorithms/src/RebinByTimeAtSample.cpp
Framework/Algorithms/src/Rebin.cpp
Framework/Algorithms/src/RebinToWorkspace.cpp
Framework/Algorithms/src/Rebunch.cpp
Framework/Algorithms/src/ReflectometryReductionOne.cpp
Framework/Algorithms/src/ReflectometryWorkflowBase.cpp
Framework/Algorithms/src/Regroup.cpp
Framework/Algorithms/src/RemoveBackground.cpp
Framework/Algorithms/src/RemoveBins.cpp

tom-perkins, 2016-10-03:
Picked the following algorithms:
```
Framework/Algorithms/src/RemoveExpDecay.cpp
Framework/Algorithms/src/RemoveLowResTOF.cpp
Framework/Algorithms/src/RemoveMaskedSpectra.cpp
Framework/Algorithms/src/ResampleX.cpp
Framework/Algorithms/src/ResetNegatives.cpp
```

DavidFair, 2016-10-03:
#17669

```Framework/Algorithms/src/RingProfile.cpp
Framework/Algorithms/src/RRFMuon.cpp
Framework/Algorithms/src/SampleCorrections/MayersSampleCorrection.cpp
Framework/Algorithms/src/SANSDirectBeamScaling.cpp
Framework/Algorithms/src/SetUncertainties.cpp```

raquelalvarezbanos, 2016-10-03:
https://github.com/mantidproject/mantid/issues/17673

`Framework/Algorithms/src/SmoothNeighbours.cpp`
`Framework/Algorithms/src/SofQWCentre.cpp`
`Framework/Algorithms/src/SofQWNormalisedPolygon.cpp`
`Framework/Algorithms/src/SofQWPolygon.cpp`
`Framework/Algorithms/src/SolidAngle.cpp`
`Framework/Algorithms/src/Stitch1D.cpp`

DimitarTasev, 2016-10-03:
Framework/Algorithms/src/StripVanadiumPeaks.cpp
Framework/Algorithms/src/SumEventsByLogValue.cpp
Framework/Algorithms/src/SumRowColumn.cpp
Framework/Algorithms/src/SumSpectra.cpp
Framework/Algorithms/src/TOFSANSResolutionByPixel.cpp

AntonPiccardoSelg, 2016-10-04:
Picking:
Framework/Algorithms/src/UnwrapMonitor.cpp
Framework/Algorithms/src/UnwrapSNS.cpp
Framework/Algorithms/src/VesuvioL1ThetaResolution.cpp
Framework/Algorithms/src/WeightedMeanOfWorkspace.cpp
Framework/Algorithms/src/WienerSmooth.cpp

DimitarTasev, 2016-10-04:
Framework/Algorithms/src/WorkspaceJoiners.cpp
Framework/Crystal/src/AnvredCorrection.cpp
Framework/Crystal/src/CentroidPeaks.cpp
Framework/Crystal/src/FindSXPeaks.cpp
Framework/Crystal/src/IntegratePeakTimeSlices.cpp

PranavBahuguna, 2016-10-05:
#17700

- Framework/Crystal/src/LoadIsawSpectrum.cpp
- Framework/Crystal/src/MaskPeaksWorkspace.cpp
- Framework/Crystal/src/NormaliseVanadium.cpp
- Framework/Crystal/src/PeakIntegration.cpp
- Framework/Crystal/src/PeakIntensityVsRadius.cpp

DavidFair, 2016-10-05:
#17705
```
Framework/Crystal/src/LoadIsawSpectrum.cpp
Framework/Crystal/src/MaskPeaksWorkspace.cpp
Framework/Crystal/src/NormaliseVanadium.cpp
Framework/Crystal/src/PeakIntegration.cpp
Framework/Crystal/src/PeakIntensityVsRadius.cpp
```

tom-perkins, 2016-10-05:
```
Framework/Crystal/src/SaveIsawPeaks.cpp
Framework/Crystal/src/SCDCalibratePanels.cpp
Framework/CurveFitting/src/Algorithms/ConvertToYSpace.cpp
Framework/CurveFitting/src/Algorithms/ConvolveWorkspaces.cpp
Framework/CurveFitting/src/Algorithms/Fit1D.cpp
```

raquelalvarezbanos, 2016-10-05:
Framework/CurveFitting/src/Algorithms/FitPowderDiffPeaks.cpp
Framework/CurveFitting/src/Algorithms/LeBailFit.cpp
Framework/CurveFitting/src/Algorithms/LeBailFunction.cpp
Framework/CurveFitting/src/Algorithms/NormaliseByPeakArea.cpp
Framework/CurveFitting/src/Algorithms/PawleyFit.cpp
Framework/CurveFitting/src/Algorithms/RefinePowderInstrumentParameters3.cpp

DimitarTasev, 2016-10-06:
```
Framework/CurveFitting/src/Algorithms/VesuvioCalculateMS.cpp
Framework/CurveFitting/src/FitMW.cpp
Framework/CurveFitting/src/FuncMinimizers/FABADAMinimizer.cpp
Framework/CurveFitting/src/FunctionDomain1DSpectrumCreator.cpp
Framework/CurveFitting/src/Functions/BivariateNormal.cpp
```

PranavBahuguna, 2016-10-06:
```
Framework/CurveFitting/src/Functions/ComptonProfile.cpp
Framework/CurveFitting/src/Functions/ComptonScatteringCountRate.cpp
Framework/CurveFitting/src/Functions/ProcessBackground.cpp
Framework/CurveFitting/src/Functions/TabulatedFunction.cpp
Framework/CurveFitting/src/HistogramDomainCreator.cpp
```

DavidFair, 2016-10-06:
```
Framework/CurveFitting/src/IMWDomainCreator.cpp
Framework/CurveFitting/src/SeqDomainSpectrumCreator.cpp
Framework/DataHandling/src/AsciiPointBase.cpp
Framework/DataHandling/src/CreateSimulationWorkspace.cpp
Framework/DataHandling/src/GroupDetectors2.cpp
```

DavidFair, 2016-10-10:
```
Framework/DataHandling/src/GroupDetectors.cpp
Framework/DataHandling/src/ImggAggregateWavelengths.cpp
Framework/DataHandling/src/LoadAscii2.cpp
Framework/DataHandling/src/LoadAscii.cpp
Framework/DataHandling/src/LoadCalFile.cpp
```

samueljackson92, 2016-10-11:
```
Framework/DataHandling/src/LoadCanSAS1D2.cpp
Framework/DataHandling/src/LoadCanSAS1D.cpp
Framework/DataHandling/src/LoadDaveGrp.cpp
Framework/DataHandling/src/LoadDetectorsGroupingFile.cpp
Framework/DataHandling/src/LoadEventPreNexus2.cpp
```

DavidFair, 2016-10-13:
```
Framework/DataHandling/src/LoadFITS.cpp
Framework/DataHandling/src/LoadGSS.cpp
Framework/DataHandling/src/LoadILLIndirect.cpp
Framework/DataHandling/src/LoadILLReflectometry.cpp
Framework/DataHandling/src/LoadILLSANS.cpp
```

PranavBahuguna, 2016-10-13:
#17808
```
Framework/DataHandling/src/LoadILLTOF.cpp
Framework/DataHandling/src/LoadISISNexus2.cpp
Framework/DataHandling/src/LoadLLB.cpp
Framework/DataHandling/src/LoadMask.cpp
Framework/DataHandling/src/LoadMcStas.cpp
```

LamarMoore, 2017-01-23:
Picked the following #18494:
Framework/DataHandling/src/LoadMcStasNexus.cpp
Framework/DataHandling/src/LoadMLZ.cpp
Framework/DataHandling/src/LoadMuonNexus1.cpp
Framework/DataHandling/src/LoadMuonNexus2.cpp
Framework/DataHandling/src/LoadNexusMonitors2.cpp
Framework/DataHandling/src/LoadNexusProcessed.cpp
Framework/DataHandling/src/LoadNXcanSAS.cpp
Framework/DataHandling/src/LoadNXSPE.cpp
Framework/DataHandling/src/LoadPDFgetNFile.cpp
Framework/DataHandling/src/LoadPreNexusMonitors.cpp

samueljackson92, 2017-01-23:
Picked the following #18495:
```
Framework/DataHandling/src/LoadQKK.cpp
Framework/DataHandling/src/LoadRawHelper.cpp
Framework/DataHandling/src/LoadRKH.cpp
Framework/DataHandling/src/LoadSassena.cpp
Framework/DataHandling/src/LoadSINQFocus.cpp
Framework/DataHandling/src/LoadSNSspec.cpp
Framework/DataHandling/src/LoadSpec.cpp
Framework/DataHandling/src/LoadSPE.cpp
Framework/DataHandling/src/LoadSpice2D.cpp
Framework/DataHandling/src/LoadSpiceXML2DDet.cpp
```

LamarMoore, 2017-01-30:
Framework/DataHandling/src/LoadTOFRawNexus.cpp
Framework/DataHandling/src/LoadVulcanCalFile.cpp
Framework/DataHandling/src/MaskDetectors.cpp
Framework/DataHandling/src/PDLoadCharacterizations.cpp
Framework/DataHandling/src/SaveAscii2.cpp
Framework/DataHandling/src/SaveAscii.cpp
Framework/DataHandling/src/SaveCanSAS1D2.cpp
Framework/DataHandling/src/SaveCanSAS1D.cpp
Framework/DataHandling/src/SaveCSV.cpp
Framework/DataHandling/src/SaveDaveGrp.cpp

DavidFair, 2017-01-30:
```
Framework/DataHandling/src/SaveDetectorsGrouping.cpp
Framework/DataHandling/src/SaveFITS.cpp
Framework/DataHandling/src/SaveFocusedXYE.cpp
Framework/DataHandling/src/SaveGSS.cpp
Framework/DataHandling/src/SaveMask.cpp
```

DimitarTasev, 2017-01-31:
#issue#
```
Framework/DataHandling/src/SaveNISTDAT.cpp
Framework/DataHandling/src/SaveNXcanSAS.cpp
Framework/DataHandling/src/SaveNXSPE.cpp
Framework/DataHandling/src/SaveNXTomo.cpp
Framework/DataHandling/src/SaveOpenGenieAscii.cpp
Framework/DataHandling/src/SavePDFGui.cpp
Framework/DataHandling/src/SaveRKH.cpp
Framework/DataHandling/src/SaveSPE.cpp
Framework/DataHandling/src/SaveToSNSHistogramNexus.cpp
Framework/DataHandling/src/SaveVTK.cpp
```

AnthonyLim23, 2017-01-31:
Framework/Examples/ModifyData.cpp
Framework/Examples/WorkspaceAlgorithm.cpp
Framework/Geometry/inc/MantidGeometry/Instrument/Component.h
Framework/Geometry/src/ComponentParser.cpp
Framework/Geometry/src/Instrument/Component.cpp

LamarMoore, 2017-02-01:
Framework/GPUAlgorithms/src/GPUTester.cpp
Framework/LiveData/src/ISIS/ISISHistoDataListener.cpp
Framework/MatlabAPI/src/MatlabInterface.cpp
Framework/MDAlgorithms/inc/MantidMDAlgorithms/AccumulateMD.h
Framework/MDAlgorithms/inc/MantidMDAlgorithms/CreateMD.h
Framework/MDAlgorithms/inc/MantidMDAlgorithms/FindPeaksMD.h
Framework/MDAlgorithms/inc/MantidMDAlgorithms/LoadSQW.h
Framework/MDAlgorithms/src/AccumulateMD.cpp
Framework/MDAlgorithms/src/ConvertCWPDMDToSpectra.cpp
Framework/MDAlgorithms/src/ConvertCWSDExpToMomentum.cpp

AnthonyLim23, 2017-02-01:
Framework/MDAlgorithms/src/ConvertMDHistoToMatrixWorkspace.cpp
Framework/MDAlgorithms/src/ConvertSpiceDataToRealSpace.cpp
Framework/MDAlgorithms/src/ConvertToMD.cpp
Framework/MDAlgorithms/src/ConvToMDHistoWS.cpp
Framework/MDAlgorithms/src/CreateMD.cpp

samueljackson92, 2017-02-01:
#18656
```
Framework/MDAlgorithms/src/DivideMD.cpp
Framework/MDAlgorithms/src/EqualToMD.cpp
Framework/MDAlgorithms/src/FindPeaksMD.cpp
Framework/MDAlgorithms/src/GetSpiceDataRawCountsFromMD.cpp
Framework/MDAlgorithms/src/GreaterThanMD.cpp
Framework/MDAlgorithms/src/IntegrateEllipsoids.cpp
Framework/MDAlgorithms/src/IntegrateFlux.cpp
Framework/MDAlgorithms/src/IntegratePeaksCWSD.cpp
Framework/MDAlgorithms/src/IntegratePeaksMD2.cpp
Framework/MDAlgorithms/src/IntegratePeaksMD.cpp
```

LamarMoore, 2017-02-02:
Framework/Nexus/src/NexusFileIO.cpp
Framework/SINQ/src/LoadFlexiNexus.cpp
Framework/SINQ/src/MDHistoToWorkspace2D.cpp
Framework/SINQ/src/PoldiAnalyseResiduals.cpp
Framework/SINQ/src/PoldiFitPeaks2D.cpp
Framework/SINQ/src/PoldiPeakSearch.cpp
Framework/SINQ/src/PoldiTruncateData.cpp
Framework/SINQ/src/PoldiUtilities/PoldiAutoCorrelationCore.cpp
Framework/SINQ/src/PoldiUtilities/PoldiResidualCorrelationCore.cpp
Framework/SINQ/src/PoldiUtilities/PoldiSpectrumConstantBackground.cpp

DavidFair, 2017-02-02:
#18684

```
Framework/SINQ/src/PoldiUtilities/PoldiSpectrumDomainFunction.cpp
Framework/SINQ/src/PoldiUtilities/PoldiSpectrumLinearBackground.cpp
Framework/TestHelpers/inc/MantidTestHelpers/WorkspaceCreationHelper.h
Framework/TestHelpers/src/MultiDomainFunctionHelper.cpp
Framework/TestHelpers/src/WorkspaceCreationHelper.cpp
```

AnthonyLim23, 2017-02-08:
Framework/WorkflowAlgorithms/src/EQSANSDarkCurrentSubtraction2.cpp
Framework/WorkflowAlgorithms/src/EQSANSMonitorTOF.cpp
Framework/WorkflowAlgorithms/src/EQSANSPatchSensitivity.cpp
Framework/WorkflowAlgorithms/src/EQSANSQ2D.cpp
Framework/WorkflowAlgorithms/src/HFIRDarkCurrentSubtraction.cpp

PranavBahuguna, 2017-02-08:
#issue

```
MantidQt/CustomInterfaces/src/Indirect/ConvFit.cpp
MantidQt/CustomInterfaces/src/Indirect/CorrectionsTab.cpp
MantidQt/CustomInterfaces/src/Indirect/IndirectDataReductionTab.cpp
MantidQt/CustomInterfaces/src/Indirect/IndirectSymmetrise.cpp
MantidQt/CustomInterfaces/src/Indirect/Iqt.cpp
MantidQt/CustomInterfaces/src/Indirect/IqtFit.cpp
MantidQt/CustomInterfaces/src/Indirect/ISISCalibration.cpp
MantidQt/CustomInterfaces/src/Indirect/ISISDiagnostics.cpp
MantidQt/CustomInterfaces/src/Indirect/ISISEnergyTransfer.cpp
MantidQt/CustomInterfaces/src/Indirect/ResNorm.cpp
```

AnthonyLim23, 2017-02-08:
MantidQt/CustomInterfaces/src/Muon/ALCBaselineModellingModel.cpp
MantidQt/CustomInterfaces/src/Muon/ALCBaselineModellingPresenter.cpp
MantidQt/CustomInterfaces/src/Muon/ALCBaselineModellingView.cpp
MantidQt/CustomInterfaces/src/Muon/ALCDataLoadingPresenter.cpp
MantidQt/CustomInterfaces/src/Muon/ALCDataLoadingView.cpp
MantidQt/CustomInterfaces/src/Muon/ALCHelper.cpp
MantidQt/CustomInterfaces/src/Muon/ALCPeakFittingPresenter.cpp
MantidQt/CustomInterfaces/src/Muon/ALCPeakFittingView.cpp
MantidQt/CustomInterfaces/src/Muon/MuonAnalysis.cpp
MantidQt/CustomInterfaces/src/Muon/MuonAnalysisFitDataPresenter.cpp
MantidQt/CustomInterfaces/src/Muon/MuonAnalysisHelper.cpp

PranavBahuguna, 2017-02-08:
#issue

```
Framework/WorkflowAlgorithms/src/RefReduction.cpp
Framework/WorkflowAlgorithms/src/RefRoi.cpp
Framework/WorkflowAlgorithms/src/SANSSolidAngleCorrection.cpp
MantidPlot/src/FunctionCurve.cpp
MantidPlot/src/Mantid/MantidGroupPlotGenerator.cpp
MantidPlot/src/Mantid/MantidMatrix.cpp
MantidPlot/src/Mantid/MantidMatrixFunction.cpp
MantidPlot/src/Mantid/MantidMatrix.h
MantidPlot/src/Mantid/MantidMatrixModel.cpp
MantidPlot/src/Mantid/MantidUI.cpp
```

PranavBahuguna, 2017-02-10:
#issue

```
MantidQt/API/src/QwtWorkspaceBinData.cpp
MantidQt/API/src/QwtWorkspaceSpectrumData.cpp
MantidQt/CustomDialogs/src/FitDialog.cpp
MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/MultiDatasetFit/MDFDatasetPlotData.h
MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/Muon/ALCBaselineModellingView.h
MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/Muon/ALCDataLoadingView.h
MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/Muon/ALCPeakFittingView.h
MantidQt/CustomInterfaces/src/DynamicPDF/DPDFDisplayControl.cpp
MantidQt/CustomInterfaces/src/DynamicPDF/DPDFFourierTransform.cpp
MantidQt/CustomInterfaces/src/DynamicPDF/DPDFInputDataControl.cpp
```

LamarMoore, 2017-02-21:
#18937

MantidQt/CustomInterfaces/src/EnggDiffraction/EnggDiffFittingPresenter.cpp
MantidQt/CustomInterfaces/src/MantidEVWorker.cpp
MantidQt/CustomInterfaces/src/MultiDatasetFit/MDFDataController.cpp
MantidQt/CustomInterfaces/src/MultiDatasetFit/MDFDatasetPlotData.cpp
MantidQt/CustomInterfaces/src/SampleTransmission.cpp
MantidQt/CustomInterfaces/src/SANSPlotSpecial.cpp
MantidQt/CustomInterfaces/src/SANSRunWindow.cpp
MantidQt/CustomInterfaces/src/Tomography/ImageROIViewQtWidget.cpp
MantidQt/CustomInterfaces/src/Tomography/ImggFormatsConvertViewQtWidget.cpp
MantidQt/CustomInterfaces/src/Tomography/TomographyIfaceViewQtGUI.cpp
MantidQt/CustomInterfaces/test/ALCBaselineModellingModelTest.h
MantidQt/CustomInterfaces/test/ALCBaselineModellingPresenterTest.h
MantidQt/CustomInterfaces/test/ALCPeakFittingModelTest.h
MantidQt/CustomInterfaces/test/MuonAnalysisDataLoaderTest.h
MantidQt/CustomInterfaces/test/MuonAnalysisHelperTest.h

SimonHeybrock, 2018-09-10:
Outdated umbrella issue.

closed 2018-09-10 by SimonHeybrock