mantidproject/mantid #17730: Add Workspace attribute to TeixeiraWaterSQE fit function

jmborr, 2016-10-06:
Same as in InelasticDiffSphere, for use with the indirect analysis interface.

**Note**: this is a temporary fix. The long-term solution is to inherit from a base-class  for fit functions that depend on `Q`. This base class would implement the setWorkspace method to retrieve the `Q` for each spectrum. This base class will implement Michael Wedel's interface `IFunction1DSpectrum` for fit functions operating on a domain of type `FunctionDomain1DSpectrum`.  when you pass the `FunctionDomain1DSpectrum` to calculate the function values, the fit function will know which value of `Q` to use thanks to the workspace index property of `FunctionDomain1DSpectrum`.

jmborr, 2016-10-06:
@spencerhowells done

SpencerHowells, 2016-10-06:
Looking forward to using it!

jmborr, 2016-12-08:
Irrelevant once #18101 is closed. Then TeixeiraWaterSQE will inherit from FunctionQDepends

closed 2016-12-08 by jmborr