mantidproject/mantid #17743: Rollout SpectrumInfo

SimonHeybrock, 2016-10-07:
Use `SpectrumInfo` in more places (wherever it makes sense).

ianbush, 2016-11-10:
The following algorithms are done in #18031:

* Framework/Algorithms/src/BinaryOperation.cpp
* Framework/Algorithms/src/CalculateTransmission.cpp
* Framework/Algorithms/src/CalculateTransmissionBeamSpreader.cpp
* Framework/Algorithms/src/ConvertSpectrumAxis.cpp
* Framework/Algorithms/src/ConvertSpectrumAxis2.cpp

ianbush, 2016-11-17:
The following algorithms are done in #18031:

* Framework/Algorithms/src/ConvertUnitsUsingDetectorTable.cpp
* Framework/Algorithms/src/CorrectKiKf.cpp
* Framework/Algorithms/src/CreatePSDBleedMask.cpp
* Framework/Algorithms/src/DetectorEfficiencyCor.cpp
* Framework/Algorithms/src/GetAllEi.cpp
* Framework/Algorithms/src/GetEi2.cpp
* Framework/Algorithms/src/GetEiMonDet2.cpp

ianbush, 2016-12-08:
The following algorithms are done in #18233:
* Framework/Algorithms/src/CalculateDIFC.cpp
* Framework/Algorithms/src/ExtractMaskToTable.cpp
* Framework/Algorithms/src/He3TubeEfficiency.cpp
* Framework/Algorithms/src/SmoothNeighbours.cpp
* Framework/Algorithms/src/SofQWNormalisedPolygon.cpp
* Framework/Algorithms/src/SampleCorrections/MayersSampleCorrection.cpp
* Framework/API/src/MatrixWorkspaceMDIterator.cpp
* Framework/CurveFitting/src/SeqDomainSpectrumCreator.cpp

SimonHeybrock, 2016-12-09:
Done in #18234:

```
Framework/DataHandling/src/GroupDetectors2.cpp
Framework/DataHandling/src/MaskDetectors.cpp
Framework/DataHandling/src/SaveGSS.cpp
Framework/DataHandling/src/SaveNXSPE.cpp
Framework/DataHandling/src/SaveSPE.cpp

Framework/DataObjects/src/ReflectometryTransform.cpp

Framework/Geometry/src/Instrument/DetectorGroup.cpp
Framework/Geometry/src/Instrument/NearestNeighbours.cpp

Framework/MDAlgorithms/src/CalculateCoverageDGS.cpp
Framework/MDAlgorithms/src/IntegrateEllipsoids.cpp
Framework/MDAlgorithms/src/IntegratePeaksMD2.cpp
Framework/MDAlgorithms/src/MDNormDirectSC.cpp
Framework/MDAlgorithms/src/MDNormSCD.cpp
Framework/MDAlgorithms/src/PreprocessDetectorsToMD.cpp

Framework/WorkflowAlgorithms/srcEQSANSDarkCurrentSubtraction2.cpp
Framework/WorkflowAlgorithms/srcEQSANSPatchSensitivity.cpp
Framework/WorkflowAlgorithms/srcSANSSolidAngleCorrection.cpp

MantidPlot/src/Mantid/MantidMatrixModel.cpp
```

ianbush, 2017-01-17:
Done in #18447 :
* Framework/Algorithms/src/HRPDSlabCanAbsorption.cpp
* ~~Framework/Algorithms/src/ModeratorTzero.cpp~~
* ~~Framework/Algorithms/src/ModeratorTzeroLinear.cpp~~
* Framework/Algorithms/src/MonteCarloAbsorption.cpp
* ~~Framework/Algorithms/src/Q1D2.cpp~~
* Framework/Algorithms/src/Qhelper.cpp
* ~~Framework/Algorithms/src/RadiusSum.cpp~~
* ~~Framework/Algorithms/src/SofQWCentre.cpp~~
* ~~Framework/Algorithms/src/SofQWPolygon.cpp~~
* ~~Framework/Algorithms/src/SpatialGrouping.cpp~~
* Framework/Algorithms/src/SpecularReflectionAlgorithm.cpp
* Framework/Algorithms/src/SumEventsByLogValue.cpp
* Framework/Algorithms/src/SumNeighbours.cpp
* ~~Framework/Algorithms/src/TimeAtSampleStrategyElastic.cpp~~
* ~~Framework/Algorithms/src/TimeAtSampleStrategyIndirect.cpp~~
* Framework/Algorithms/src/UnwrapMonitorsInTOF.cpp
* Framework/Algorithms/src/VesuvioL1ThetaResolution.cpp
* Framework/Algorithms/test/ApplyCalibrationTest.h
* Framework/Algorithms/test/CopyInstrumentParametersTest.h
* Framework/Algorithms/test/ResizeRectangularDetectorTest.h
* Framework/Algorithms/test/TimeAtSampleStrategyDirectTest.h
* Framework/Algorithms/test/TimeAtSampleStrategyElasticTest.h

Note - algorithms that are crossed out have already been updated.

OwenArnold, 2017-01-24:
I've taken 

```
Framework/Algorithms/src/AddPeak.cpp
Framework/Algorithms/src/ConvertDiffCal.cpp
Framework/Algorithms/src/ConvertSpectrumAxis.cpp
Framework/Algorithms/src/ConvertSpectrumAxis2.cpp
Framework/Algorithms/src/ConvertUnitsUsingDetectorTable.cpp
Framework/Algorithms/src/CreateLogTimeCorrection.cpp
Framework/Algorithms/src/DetectorDiagnostic.cpp
Framework/Algorithms/src/DetectorEfficiencyCor.cpp
Framework/Algorithms/src/IntegrateByComponent.cpp
Framework/Algorithms/src/NormaliseByDetector.cpp
Framework/Algorithms/src/RemoveLowResTOF.cpp
Framework/Algorithms/src/SpecularReflectionAlgorithm.cpp
Framework/Algorithms/src/SofQWPolygon.cpp
Framework/Algorithms/src/ScaleX.cpp

```

PranavBahuguna, 2017-01-25:
Doing the following in #18539:

```
Framework/Crystal/src/IntegratePeakTimeSlices.cpp
Framework/Crystal/src/FindSXPeaks.cpp
Framework/Crystal/src/AnvredCorrection.cpp
Framework/Crystal/src/PredictPeaks.cpp
Framework/Crystal/src/NormaliseVanadium.cpp
```

soininen, 2017-01-25:
I have picked:
```
Framework/API/src/IFunction.cpp
Framework/Algorithms/test/ModeratorTzeroTest.h
Framework/Algorithms/test/NormaliseByDetectorTest.h
Framework/Algorithms/test/SetInstrumentParameterTest.h
Framework/Algorithms/test/ScaleXTest.h
Framework/Algorithms/test/RingProfileTest.h
Framework/Crystal/test/AddPeakHKLTest.h
```

AntonPiccardoSelg, 2017-01-26:
I have picked:
```
Framework/API/src/IFunction.cpp
Framework/Algorithms/test/NormaliseByDetectorTest.h
Framework/Algorithms/test/SetInstrumentParameterTest.h
Framework/Algorithms/test/RingProfileTest.h
Framework/Algorithms/test/ScaleXTest.h
```

ianbush, 2017-01-26:
I have picked:
```
Framework/Crystal/test/AddPeakHKLTest.h
Framework/Crystal/test/PeaksInRegionTest.h
Framework/Crystal/test/IntegratePeakTimeSlicesTest.h
Framework/Crystal/src/IntegratePeakTimeSlices.cpp
Framework/Crystal/src/FindSXPeaks.cpp
Framework/Crystal/src/PredictPeaks.cpp
Framework/CurveFitting/src/Functions/ComptonProfile.cpp
Framework/CurveFitting/src/Functions/InelasticDiffSphere.cpp
Framework/CurveFitting/src/Functions/InelasticDiffRotDiscreteCircle.cpp
Framework/CurveFitting/test/Algorithms/VesuvioCalculateMSTest.h
Framework/CurveFitting/test/Functions/ComptonProfileTestHelpers.h
Framework/CurveFitting/test/FitMWTest.h
```
for #18567

raquelalvarezbanos, 2017-01-26:
I have picked:

```
Framework/DataHandling/src/LoadHelper.cpp
Framework/DataHandling/src/LoadInstrumentFromNexus.cpp
Framework/DataHandling/src/SaveGSS.cpp
Framework/DataHandling/src/LoadVulcanCalFile.cpp
Framework/DataHandling/src/MoveInstrumentComponent.cpp
Framework/DataHandling/src/RotateInstrumentComponent.cpp
Framework/DataHandling/src/SaveAscii2.cpp
Framework/DataHandling/src/SaveFocusedXYE.cpp
Framework/DataHandling/test/LoadDetectorInfoTest.h
Framework/DataHandling/test/LoadEmptyInstrumentTest.h
Framework/DataHandling/test/GroupDetectors2Test.h
Framework/DataHandling/test/LoadIDFFromNexusTest.h
Framework/DataHandling/test/LoadISISNexusTest.h
Framework/DataHandling/test/LoadInstrumentFromRawTest.h
Framework/DataHandling/test/LoadInstrumentTest.h
```

in https://github.com/mantidproject/mantid/issues/18569

AntonPiccardoSelg, 2017-01-26:
I have picked:

```
Framework/DataHandling/.../LoadMaskTest.h
Framework/DataHandling/.../LoadGSSTest.h
Framework/DataHandling/.../LoadMuonNexus2Test.h
Framework/DataHandling/.../LoadParameterFileTest.h
Framework/DataHandling/.../LoadQKKTest.h
Framework/DataHandling/.../LoadRaw3Test.h
Framework/DataHandling/.../LoadRawSaveNxsLoadNxsTest.h
Framework/DataHandling/.../LoadSpiceXML2DDetTest.h
Framework/DataHandling/.../MaskDetectorsTest.h
Framework/DataHandling/.../SaveParameterFileTest.h
```

AntonPiccardoSelg, 2017-01-27:
I have picked:

```
Framework/DataHandling/.../SetScalingPSDTest.h
Framework/DataHandling/.../UpdateInstrumentFromFileTest.h
Framework/DataObjects/.../EventWorkspace.cpp
Framework/DataObjects/.../Peak.cpp
Framework/DataObjects/.../PeakTest.h
Framework/DataObjects/.../PeaksWorkspaceTest.h
Framework/DataObjects/.../ReflectometryTransformTest.h
Framework/DataObjects/.../Workspace2DTest.h
Framework/DataObjects/inc/MantidDataObjects/Peak.h
Framework/Geometry/.../ComponentHelperTest.h
```

PranavBahuguna, 2017-01-27:
I have picked:

```
Framework/MDAlgorithms/.../ConvertCWSDExpToMomentum.cpp
Framework/MDAlgorithms/.../ConvertSpiceDataToRealSpace.cpp
Framework/MDAlgorithms/.../ConvertToDiffractionMDWorkspace.cpp
Framework/MDAlgorithms/.../ConvertToMDMinMaxGlobal.cpp
Framework/MDAlgorithms/.../IntegratePeaksCWSD.cpp
Framework/MDAlgorithms/.../LoadILLAscii.cpp
Framework/WorkflowAlgorithms/.../DgsConvertToEnergyTransfer.cpp
Framework/WorkflowAlgorithms/.../EQSANSLoad.cpp
Framework/WorkflowAlgorithms/.../EQSANSMonitorTOF.cpp
Framework/WorkflowAlgorithms/.../RefReduction.cpp
```

DavidFair, 2017-01-30:
**Group 1 - @DavidFair:**
```
Framework/Geometry/.../Instrument/InstrumentDefinitionParser.cpp
Framework/Geometry/.../InstrumentDefinitionParserTest.h
Framework/Geometry/.../ParComponentFactoryTest.h
Framework/Geometry/.../ParInstrumentTest.h
Framework/Geometry/inc/MantidGeometry/Crystal/IPeak.h
Framework/MDAlgorithms/.../CalculateCoverageDGSTest.h
```

AntonPiccardoSelg, 2017-01-30:
I picked:

**Group 3**
```
MantidQt/MantidWidgets/.../InstrumentView/ComponentActor.cpp
MantidQt/MantidWidgets/.../InstrumentView/InstrumentActor.cpp
MantidQt/MantidWidgets/.../InstrumentView/InstrumentWidgetMaskTab.cpp
MantidQt/MantidWidgets/.../InstrumentView/InstrumentWidgetPickTab.cpp
MantidQt/MantidWidgets/.../InstrumentView/ObjComponentActor.cpp
MantidQt/MantidWidgets/.../InstrumentView/ProjectionSurface.cpp
MantidQt/MantidWidgets/.../InstrumentView/RotationSurface.cpp
```

gemmaguest, 2017-01-30:
I have picked:

**Group 6:**
```
Framework/Algorithms/src/DetectorEfficiencyCor.cpp
Framework/Algorithms/src/IntegrateByComponent.cpp
Framework/Algorithms/src/NormaliseByDetector.cpp
Framework/Algorithms/src/RemoveLowResTOF.cpp
Framework/Algorithms/src/SpecularReflectionAlgorithm.cpp
Framework/Algorithms/src/SofQWPolygon.cpp
Framework/Algorithms/src/ScaleX.cpp
```

AntonPiccardoSelg, 2017-01-30:
I have picked:

**Group 4 - @AntonPiccardoSelg:**
```
MantidQt/MantidWidgets/inc/MantidQtMantidWidgets/InstrumentView/ComponentActor.h
MantidQt/MantidWidgets/inc/MantidQtMantidWidgets/InstrumentView/InstrumentActor.h
MantidQt/MantidWidgets/inc/MantidQtMantidWidgets/InstrumentView/ProjectionSurface.h
```

**Group 5 - @AntonPiccardoSelg:**
```
MantidQt/SliceViewer/.../QpeaksTableModel.cpp
```

KarlPalmen, 2017-01-30:
Have all files that need changing been taken?

gemmaguest, 2017-01-30:
@KarlPalmen Yes, I think so.

ianbush, 2017-01-31:
@KarlPalmen Yes, just to confirm all files that can be updated now have been taken. Thanks all!

ianbush, 2017-01-31:
@KarlPalmen @gemmaguest - just found one more group to add :smile: .

PranavBahuguna, 2017-02-02:
I have picked:

**Group 7 - @PranavBahuguna**

```
Framework/Algorithms/src/ConvertToConstantL2.cpp
Framework/API/test/MatrixWorkspaceTest.h
Framework/DataHandling/src/LoadDetectorInfo.cpp
Framework/DataHandling/src/LoadHelper.cpp
Framework/DataHandling/src/LoadILLIndirect.cpp
Framework/DataHandling/src/LoadILLIndirect2.cpp
Framework/DataHandling/src/LoadIsawDetCal.cpp
Framework/DataHandling/src/MoveInstrumentComponent.cpp
Framework/DataHandling/src/RotateInstrumentComponent.cpp
Framework/DataHandling/src/SetScalingPSD.cpp1
```

in #issue

ianbush, 2017-02-02:
Assigned to @OwenArnold in #18687:
```
Framework/DataObjects/.../Peak.cpp
Framework/DataObjects/.../PeakTest.h
Framework/DataObjects/.../PeaksWorkspaceTest.h
```

ianbush, 2017-02-02:
Taken in #18688:
```
MantidQt/MantidWidgets/.../InstrumentView/ComponentActor.cpp
MantidQt/MantidWidgets/.../InstrumentView/InstrumentActor.cpp
MantidQt/MantidWidgets/.../InstrumentView/InstrumentWidgetPickTab.cpp
MantidQt/MantidWidgets/.../InstrumentView/ObjComponentActor.cpp
MantidQt/MantidWidgets/.../InstrumentView/ProjectionSurface.cpp
MantidQt/MantidWidgets/.../InstrumentView/RotationSurface.cpp
MantidQt/MantidWidgets/inc/MantidQtMantidWidgets/InstrumentView/ComponentActor.h
MantidQt/MantidWidgets/inc/MantidQtMantidWidgets/InstrumentView/InstrumentActor.h
MantidQt/MantidWidgets/inc/MantidQtMantidWidgets/InstrumentView/ProjectionSurface.h
```

SimonHeybrock, 2017-02-28:
I am closing this umbrella issue since basically everything has been dealt with. A few remaining cases are dealt with in https://github.com/mantidproject/mantid/pull/18906 or will need special attention at a later point.

Thanks to everyone who contributed!

closed 2017-02-28 by SimonHeybrock