mantidproject/mantid #17831: Umbrella: Powder diffraction instruments migration to PEARL scripts

DavidFair, 2016-10-14:
This umbrella issue holds the high level steps needed for the migration of powder diffraction instruments at ISIS to the PEARL powder diffraction script.

 Prep work:
- [ ] Move duplicated code (e.g. debug steps) into separate functions
- [ ] Remove use of globals/hardcoded values where possible
- [ ] Migrate functions so there is a hierarchy of calls with most abstract on top and the calls to Mantid API's are wrapped where possible

 POLARIS:
- [ ] Support and load options in current pref files - decide on new format to make it a bit more robust to errors in it.  
- [ ] Replicate / Tidy existing workflow from existing scripts and migrate them into new script
- [ ] As part of that figure out and isolate instrument specific steps from main steps. For example PEARL uses anvils for their samples and will divide by a pre-calculated file to offset this. This doesn't apply to POLARIS
- [ ] Change the grouping input file so its easily distinguished from the automatically generated offset cal fine. 
- [ ] Ron tells me the script currently attempts to compensate for noisy/quiet detectors which can cause issues with making faulty quiet detectors loud or masking valid data. Look into this and possibly remove the logic (as they mask faulty detectors in the grouping cal file themselves)
- [ ] Support the custom syntax of selecting runs. I'm sure Load in Mantid handles what we need specifically `1-12-3` loads files as groups of `1-3` `4-6` `7-9`...etc. 
- [ ] As an aside - `LowerLambda` and `UpperLambda` are specified in the file. These are specific to an instrument but Ron wasn't sure what they did. If I could figure out in the scripts what they do this would also make future migrations easier.

DavidFair, 2017-01-27:
As this is an ongoing process moving the ticket to 3.10

Anders-Markvardsen, 2018-03-29:
From discussion with @josephframsay and @DanNixon this issue is out-of-date and requirements and priorities have changed since this issue was last worked on

closed 2018-03-29 by Anders-Markvardsen