mantidproject/mantid #17870: Read Selected Information from Saved ParameterMap

OwenArnold, 2016-10-21:
Eventually we need to be able to move data from ParameterMap to SpectrumInfo/DetectorInfo and vice versa.

In terms of scanning instruments, we need to have positions and rotations stored in the DetectorInfo. We need a way to be able to read this information out of the ParameterMap (and resave it when required).

SimonHeybrock, 2016-12-12:
After having looking at this more closely I am not sure we actually need to interact with the ParameterMap, since it does not seem to be saved directly. More details are described here: https://github.com/mantidproject/mantid/issues/18228.

SimonHeybrock, 2016-12-14:
I think this is really covered by some of the other issues we have opened and partially already dealt with. I am closing this, please reopen if you think this is not the case.

closed 2016-12-14 by SimonHeybrock

OwenArnold, 2016-12-14:
@SimonHeybrock. It makes to close this.