mantidproject/mantid #17894: MantidUI Interface Breakdown

LamarMoore, 2016-10-27:
This comes from the work associated with #17315  where the WorkspaceDock was redesigned using the MVP pattern. An interface to MantidUI called MantidDisplayBase was created which contained all the methods required by the WorkspaceDock. It would be nice to split up the responsibilites using something similar to the mixin architecture.

OwenArnold, 2016-10-27:
Examples suggestions might be things like `PlotableWorkspaceView`, `SaveableWorkspaceView`. We can talk about this at the time of implementation.

martyngigg, 2016-10-27:
This all sounds very valuable in improving the general maintenance of the MantidPlot code.

There are currently early discussions around the future of MantidPlot itself. Ideas are floating around about writing a new python-based workbench that would becomes MantidPlot's successor but this is by no means certain.

I would therefore advise not putting too much effort into this type of refactoring work until these decisions are more concrete - it would be shame to waste the effort.

OwenArnold, 2016-10-27:
@martyngigg Good point.

martyngigg, 2016-10-27:
These ideas should still be transferable though, if a new application is the decided route .

closed 2021-03-03 by stale[bot]