mantidproject/mantid #17907: WorkspaceFactory fail when creating a workspace from parent with 1 histogram and TextAxis

DimitarTasev, 2016-10-31:
### Expected behavior
`WorkspaceFactory.create(<parent>..)` produces a usable workspace inheriting the parent's meta data

### Actual behavior
When using `WorkspaceFactory.create(...)` on a workspace with TextAxis and 1 histogram (I haven't tested if it breaks with multiple histograms) the produced workspace causes an unexpected crash in Mantid, and makes that workspace unusable.

### Steps to reproduce the behavior
Run the following script
```
dataX = []
dataY = []
YVals = ["Text"]
for x in range(20):
    dataX.append(x)
    dataY.append(x)

print dataX
print dataY
print YVals

we = CreateWorkspace(dataX, dataY, UnitX="Wavelength", \
VerticalAxisUnit="Text", VerticalAxisValues=YVals, YUnitLabel="Counts", \
WorkspaceTitle="workspacetitle")

# inherit from parent workspace, and copy the histogram 5 times
third_ws = WorkspaceFactory.create(we, NVectors=5)
mtd['labelledworkspace']=third_ws
```

### Platforms affected
Tested only on Ubuntu 14.04, but probably affects all

DTasev, 2017-06-30:
Possibly related to work in https://github.com/mantidproject/mantid/issues/17902

DTasev, 2019-04-08:
Issue was looked at into #20501, but the PR was closed without merging. Closing issue as well

closed 2019-04-08 by DTasev