mantidproject/mantid #17955: Double clicking in the workspace window causes Mantid to crash

lottiegreenwood, 2016-11-03:
### Expected behavior
Double clicking within the workspace windows should not cause Mantid to crash

### Actual behavior
Double clicking within the workspace window, however, does cause Mantid to crash

In debug it states a debug assertation has failed, and a vector subscript is out of range.

### Steps to reproduce the behavior
Build master in debug and double click within the workspace window

### Platforms affected
Windows 7, occurs on the latest nightly release

NickDraper, 2016-11-03:
Martyn can you have a look at this please

martyngigg, 2016-11-03:
I have moved this back to v3.9 as this is a recently introduced bug in https://github.com/mantidproject/mantid/pull/17339 when refactoring the widget.

PR #17965 (2016-11-07): Fix crash on double click with no selection in workspace dock
Adds a check that a selection exists before trying to import anything into the MantidPlot window.

**To test:**

### Test that the crash no longer occurs

* Start MantidPlot
* Double-click in the white space within the "workspaces" widget - nothing should happen

### Test old behaviour is the same

Using an installed copy of v3.8 compare that the behaviour is the same when double clicking on a workspace in the dock. This should include all workspace types including groups.

Fixes #17955.

*Does not need to be in the release notes* - [... cut]
closed 2016-11-07 by AndreiSavici