mantidproject/mantid #17989: Using the spectrum viewer with workspace of NAN values results in bad alloc

DavidFair, 2016-11-07:
### Expected behavior
The spectrum viewer should gracefully handle a workspace being opened with only `nan` values present

### Actual behavior
Mantid throws `bad alloc` when opening the workspace in the spectrum viewer. I'm not sure if its if there is any `nan` values or if the workspace has no data only special values (in this case)

### Steps to reproduce the behavior
Open a workspace full of `nan` values. (I have one saved however it is 300MB so I'll send it on request)
Attempt to open the spectrum viewer on that workspace.
Observe exception is thrown

### Platforms affected
All

soininen, 2016-12-16:
This also happens when opening the spectrum viewer for the workspace produced by the following script:
```python
import numpy

ws = LoadEmptyInstrument(InstrumentName='IN4')
ws.dataY(1)[0] = numpy.inf
```
So a single `inf` is enough to trigger the crash.

soininen, 2016-12-16:
@DavidFair Could you confirm that the `bad_alloc` still exists with your workspace? I tried to reproduce the issue with the script
```python
import numpy

empty_in4 = LoadEmptyInstrument(InstrumentName='IN4')
bins = numpy.arange(0, 1, 0.1)
xs = numpy.repeat(bins, empty_in4.getNumberHistograms())
ys = numpy.repeat(numpy.repeat(numpy.nan, len(bins) - 1), empty_in4.getNumberHistograms())
ws_nan = CreateWorkspace(DataX=xs, DataY=ys, NSpec=empty_in4.getNumberHistograms(), ParentWorkspace=empty_in4)
```
which gives me a workspace full of `nan`s, but didn't see the `bad_alloc` in Spectrum Viewer.

DavidFair, 2017-05-18:
This issue has been resolved with my workspace now in the latest `master`. For that reason I'll close this issue as fixed.

closed 2017-05-18 by DavidFair