mantidproject/mantid #18003: Add documentation and rename Damping minimizer

raquelalvarezbanos, 2016-11-08:
This issue is to rename Damping minimizer to Damped Gauss-Newton and add an .rst file for it.

- As a first step, ask Roman if he sees any reason why Damping should not be renamed to "Damped Gauss-Newton".
- If he agrees with the new name:
  - We need to update: http://docs.mantidproject.org/nightly/concepts/FittingMinimizers.html
  - Check with Roman that the link to Gauss-Newton algorithm in the above .html file is correct.
  - We need to manually update the name in the function browser.

KarlPalmen, 2016-11-16:
Issue #15014 will provide a stub document for this. I had difficulty finding the code for this, because it did not have GaussNewton in the name and I had to ask @mantid-roman to identify it.

KarlPalmen, 2016-11-23:
Document is now found at http://docs.mantidproject.org/nightly/fitminimizers/GaussNewton.html arising from the merger of #15014 .

PR #18146 (2016-12-07): Document and Rename Damping Minimizer to Damped Gauss-Newton
This is part of the documentation and improvement of fitting minimizers. 
This will be mentioned in the release notes later on.

**To test:**
Build HTML documentation.
Check the documentation to find the documentation for DampedGaussNewtonMinimizer exists, 
the Damping minimizer has been renamed to this and 
other occurrences of "Damping"  have not been accidentally changed.

Check all documents that may have been modified by these changes.

Also build QTHelp and check the Mantid help.

Check that the DampedGaussNewtonMinimizer can b [... cut]
closed 2016-12-07 by martyngigg