mantidproject/mantid #18004: Add documentation for Trust region minimizer

raquelalvarezbanos, 2016-11-08:
This issue is to add an .rst file for the Trust region minimizer.

- Before doing any work on this, ask Anders about this issue.

Anders-Markvardsen, 2016-12-06:
Tyrone from SCD Numerical Group suggests we describe the trust region minimizer as:

This minimizer is a hybrid Gauss-Newton_ /Quasi-Newton_ method, solved within a trust-region_.  

It is a re-implementation of (a subset of the functionality of) the RALFit_ nonlinear least-squares solver.

.. _Gauss-Newton https://en.wikipedia.org/wiki/Gauss%E2%80%93Newton_algorithm
.. _Quasi-Newton https://en.wikipedia.org/wiki/Quasi-Newton_method
.. _trust-region https://en.wikipedia.org/wiki/Trust_region
.. _RALFit https://github.com/ralna/RALFit

KarlPalmen, 2016-12-06:
Thank you for the information.

PR #18209 (2016-12-12): Basic document for trust region minimizer
Description of work.

Created a basic document for the Trust Region Minimizer.

Fixes #18004 .

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with small focussed classes/methods/functions?
- [ ] Are there unit/system tests in place? Are the unit tests small and test the a cl [... cut]
closed 2016-12-12 by SimonHeybrock