mantidproject/mantid #18033: Performance Regression in MatrixWorkspace

LamarMoore, 2016-11-10:
During a survey of the performance tests I noticed a regression in the following two tests in MatrixWorkspaceTestPerformance:

test_move_bank_and_read_positions_SpectrumInfo_x10
test_rotate_bank_and_read_positions_SpectrumInfo_x10

Which seem to be related to changes made by @SimonHeybrock  [here](https://github.com/mantidproject/mantid/pull/17875)

SimonHeybrock, 2016-12-02:
I had a short look. I did expect a small slowdown due to the way `SpectrumInfo` was refactored to now use `DetectorInfo`. The factor 2x seems a bit large> http://builds.mantidproject.org/job/master_performancetests/Master_branch_performance_tests/APITest.MatrixWorkspaceTestPerformance.test_move_bank_and_read_positions_SpectrumInfo_x10.htm.

This will be refactored several more times in the near future (and thus get much faster), so I think the priority for fixing this is low.

SimonHeybrock, 2016-12-02:
But it *is* really weird. I'll dig into it some more.

SimonHeybrock, 2016-12-21:
I built and tested several versions of this, my results are:

```
7fc8878 (2016-11-01, before jump):

$ cat ./bin/Testing/TEST-APITest.MatrixWorkspaceTestPerformance.xml
<?xml version="1.0" encoding="UTF-8" ?>
<testsuite name="APITest"  tests="7" errors="0" failures="0" package="APITest" time="3.73672" >
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_typical" line="1444" time="0.026973" totalTime="0.027038" CPUFraction="0.999963"> <measurement><name>CPUFraction</name><value>0.999963</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_calculateL2" line="1460" time="0.085354" totalTime="0.085375" CPUFraction="0.999918"> <measurement><name>CPUFraction</name><value>0.999918</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_calculateL2_x10" line="1477" time="0.810802" totalTime="0.810834" CPUFraction="0.999874"> <measurement><name>CPUFraction</name><value>0.999874</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_rotate_bank_and_read_positions_x10" line="1503" time="0.778772" totalTime="0.778799" CPUFraction="0.999914"> <measurement><name>CPUFraction</name><value>0.999914</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_move_bank_and_read_positions_x10" line="1529" time="0.752366" totalTime="0.752398" CPUFraction="0.999902"> <measurement><name>CPUFraction</name><value>0.999902</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_rotate_bank_and_read_positions_SpectrumInfo_x10" line="1552" time="0.651899" totalTime="0.651936" CPUFraction="0.99992"> <measurement><name>CPUFraction</name><value>0.99992</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_move_bank_and_read_positions_SpectrumInfo_x10" line="1570" time="0.630313" totalTime="0.630341" CPUFraction="0.99991"> <measurement><name>CPUFraction</name><value>0.99991</value></measurement>    </testcase>
</testsuite>

47a1cb1 (2016-11-04, after jump):

$ cat ./bin/Testing/TEST-APITest.MatrixWorkspaceTestPerformance.xml
<?xml version="1.0" encoding="UTF-8" ?>
<testsuite name="APITest"  tests="7" errors="0" failures="0" package="APITest" time="3.82489" >
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_typical" line="1439" time="0.025863" totalTime="0.025923" CPUFraction="0.999884"> <measurement><name>CPUFraction</name><value>0.999884</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_calculateL2" line="1455" time="0.083921" totalTime="0.083939" CPUFraction="0.999917"> <measurement><name>CPUFraction</name><value>0.999917</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_calculateL2_x10" line="1472" time="0.804965" totalTime="0.805005" CPUFraction="0.999914"> <measurement><name>CPUFraction</name><value>0.999914</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_rotate_bank_and_read_positions_x10" line="1498" time="0.767954" totalTime="0.767983" CPUFraction="0.999913"> <measurement><name>CPUFraction</name><value>0.999913</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_move_bank_and_read_positions_x10" line="1524" time="0.733023" totalTime="0.733051" CPUFraction="0.9999"> <measurement><name>CPUFraction</name><value>0.9999</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_rotate_bank_and_read_positions_SpectrumInfo_x10" line="1547" time="0.699041" totalTime="0.699068" CPUFraction="0.99992"> <measurement><name>CPUFraction</name><value>0.99992</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_move_bank_and_read_positions_SpectrumInfo_x10" line="1565" time="0.709889" totalTime="0.709923" CPUFraction="0.999911"> <measurement><name>CPUFraction</name><value>0.999911</value></measurement>    </testcase>
</testsuite>

faf0a47 (recent master):

$ cat ./bin/Testing/TEST-APITest.MatrixWorkspaceTestPerformance.xml
<?xml version="1.0" encoding="UTF-8" ?>
<testsuite name="APITest"  tests="7" errors="0" failures="0" package="APITest" time="4.04261" >
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_typical" line="1425" time="0.029681" totalTime="0.029758" CPUFraction="0.999933"> <measurement><name>CPUFraction</name><value>0.999933</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_calculateL2" line="1441" time="0.092343" totalTime="0.09237" CPUFraction="0.999913"> <measurement><name>CPUFraction</name><value>0.999913</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_calculateL2_x10" line="1458" time="0.932413" totalTime="0.932452" CPUFraction="0.99991"> <measurement><name>CPUFraction</name><value>0.99991</value></measurement>    </testcase>
[... cut ...]
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_rotate_bank_and_read_positions_SpectrumInfo_x10" line="1533" time="0.70605" totalTime="0.706101" CPUFraction="0.999908"> <measurement><name>CPUFraction</name><value>0.999908</value></measurement>    </testcase>
    <testcase classname="APITest.MatrixWorkspaceTestPerformance" name="test_move_bank_and_read_positions_SpectrumInfo_x10" line="1551" time="0.667216" totalTime="0.667251" CPUFraction="0.999921"> <measurement><name>CPUFraction</name><value>0.999921</value></measurement>    </testcase>
</testsuite>
```

ianbush, 2018-06-28:
Closing this, the performance tests now show big improvements after implementing the real `DetectorInfo` and `SpectrumInfo` classes.

closed 2018-06-28 by ianbush